CameraKit/camerakit-android #157: Portrait mode having issue

vishalpatel1327, 2017-08-25:
Hello @dwillmc   I just clone your code   becase i need to change bit rate also 

after your changes of screen orientation at this #111 

What i show that front camera  portrait mode   rotate 180 degree. :( 
but when i used  

> compile 'com.flurgle:camerakit:0.9.18

Its work fine 

[image]
[image]

Ninjaman494, 2017-08-28:
+1 Having the same issue with pulled code. Was working fine with 0.9.18

Ninjaman494, 2017-08-28:
I found a solution, change the following line in the calculateCaptureRotation method of the Camera1 class:
` return (previewRotation + 2*mDisplayOrientation + 720) %360;`

to:
` return (previewRotation + 2*mDisplayOrientation + 180) %360;`

I've only tested this on my Samsung Galaxy S4, please try it out to see if it works on other devices.

closed 2018-02-03 by austinkettner