CameraKit/camerakit-android #158: Fail to connect to camera service

jondoes, 2017-08-25:
`FATAL EXCEPTION: CameraViewWorker
                                                             Process: com.tot.app, PID: 30395
                                                             java.lang.RuntimeException: Fail to connect to camera service
                                                                 at android.hardware.Camera.<init>(Camera.java:497)
                                                                 at android.hardware.Camera.open(Camera.java:342)
                                                                 at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:378)
                                                                 at com.flurgle.camerakit.Camera1.start(Camera1.java:96)
                                                                 at com.flurgle.camerakit.CameraView$3.run(CameraView.java:302)
                                                                 at android.os.Handler.handleCallback(Handler.java:739)
                                                                 at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                 at android.os.Looper.loop(Looper.java:135)
                                                                 at android.os.HandlerThread.run(HandlerThread.java:61)`

jondoes, 2017-08-25:
5 19:17:35.800 19929-19949/com.abc.app E/BufferQueueProducer: [unnamed-19929-1] dequeueBuffer: BufferQueue has been abandoned

Pear0, 2017-09-06:
What device does this happen on?

devoma, 2017-11-17:
I have a same problem. Device LG K8, Android 6 M

Xenero, 2017-12-07:
Happening here still on a Samsung Galaxy S7 Edge.

If it is of any help it happens only when I return to the page that the camera is on after accessing a camera on another page. Note that in all the scenarios below, I have properly started and stopped the camera on the initial activity inside `onPause();` and `onResume();`.

**Scenario 1 - User receives in-app video call**
1. Application opens up to camera page
2. User receives video call and accepts which takes them to a different activity
3. Call is terminated and `finsh();` is called which returns user to the initial camera page
4. App throws error and crashes

**Scenario 2 - User access camera for any reason in another Activity/Fragment**
1. Application opens up to camera page
2. User browses through app and, for example, decides to change their profile pic by taking a photo (video also throws the issue) from the built in camera
3. User successfully uploads the new image/video and continues their journey through the app
4. User eventually returns to initial Activity at which point the error is thrown and app crashes

Pear0, 2017-12-07:
I believe Android apps gain an exclusive lock on the camera while they're using it. This is pure speculation, but my guess is that CameraKit is trying to initialize the camera before the other activity has deconstructed it. 

Right now I think CameraKit waits 100ms, but if it is that race condition, maybe CameraKit needs to wait longer or preferably do something to detect if it can initialize the camera.

Danm72, 2018-02-27:
Did you guys figure anything out here?

austinkettner, 2018-02-27:
Can you test this on the 1.0.0 PR here: #318 -- this issue should be resolved. We are heavily focused on getting 1.0.0 shipped so the goal is integrate all testing around that, if this issue still exists, add a comment in the PR please! =)

Closing as I believe this to be resolved in 1.0.0, but any PR comments will be addressed rapidly!

@Danm72 @jondoes @Pear0 @Xenero

closed 2018-02-27 by austinkettner

nandantal, 2018-05-23:
Same problem on 0.13.1

Fatal Exception: java.lang.RuntimeException: Fail to connect to camera service
       at android.hardware.Camera.<init>(Camera.java:495)
       at android.hardware.Camera.open(Camera.java:341)
       at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:667)
       at com.wonderkiln.camerakit.Camera1.start(Camera1.java:124)
       at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:266)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)

Device LG, Android 6.0.1

nandantal, 2018-05-23:
Same problem on Android 6.0.1 , LG device 

Fatal Exception: java.lang.RuntimeException: Fail to connect to camera service
       at android.hardware.Camera.<init>(Camera.java:495)
       at android.hardware.Camera.open(Camera.java:341)
       at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:667)
       at com.wonderkiln.camerakit.Camera1.start(Camera1.java:124)
       at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:266)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)

Xenero, 2018-05-24:
@austinkettner how have you closed this when it is clearly not resolved? There has been no 1.0.0 release so this needs to be left open for all suffering from this issue to have a place to report.

nandantal, 2018-05-28:
The previous problem seems like fixed with 1.0.0 , but I am getting the following error now : 

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference
       at com.camerakit.CameraKitView$Camera2$1$3.run(CameraKitView.java:1737)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)

Android 6.0.1 LG

Xenero, 2018-05-28:
Please show us your full log so we can trace what happened before this
error.