CameraKit/camerakit-android #159: java.lang.RuntimeException: setParameters failed

zhangyaboPersion, 2017-08-28:
java.lang.RuntimeException: setParameters failed
                                                                                                      at android.hardware.Camera.native_setParameters(Native Method)
                                                                                                      at android.hardware.Camera.setParameters(Camera.java:3028)
                                                                                                      at com.flurgle.camerakit.Camera1$4.onTouch(Camera1.java:515)
                                                                                                      at android.view.View.dispatchTouchEvent(View.java:9422)
                                                                                                      at com.flurgle.camerakit.CameraView$2.onTouch(CameraView.java:131)

cotfas, 2017-08-28:
It happened to me also

                                                   java.lang.RuntimeException: java.io.IOException: setPreviewTexture failed
                                                       at com.flurgle.camerakit.Camera1.setupPreview(Camera1.java:336)
                                                       at com.flurgle.camerakit.Camera1.access$100(Camera1.java:30)
                                                       at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:74)
                                                       at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
                                                       at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:80)
                                                       at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:26)
                                                       at android.view.TextureView.getHardwareLayer(TextureView.java:387)
                                                       at android.view.TextureView.draw(TextureView.java:325)
                                                       at android.view.View.updateDisplayListIfDirty(View.java:16159)
                                                       at android.view.View.draw(View.java:16943)
                                                       at android.view.ViewGroup.drawChild(ViewGroup.java:3776)

jaksab, 2017-09-02:
I get this bug also цhen I click on the left side of the screen.
It appears when the focus mode is FOCUS_TAP.

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc

Diederikjh, 2017-12-04:
Getting this crash on 0.12.1.  (As seen in [this repo](https://github.com/Diederikjh/camerakit-crash).)  On device: [SM-T561](https://www.gsmarena.com/samsung_galaxy_tab_e_9_6-7295.php)
Steps to reproduce:

1. Launch activity in linked repo.
2. Touch to focus.
3. Wait about 2 seconds.
4. Crash with below logs:

```
12-04 16:04:14.461 21173-21173/com.honeybeeapp.camerakittesting D/AndroidRuntime: Shutting down VM
12-04 16:04:14.461 21173-21173/com.honeybeeapp.camerakittesting W/dalvikvm: threadid=1: thread exiting with uncaught exception (group=0x41718bc0)
12-04 16:04:14.461 21173-21173/com.honeybeeapp.camerakittesting E/AndroidRuntime: FATAL EXCEPTION: main
  Process: com.honeybeeapp.camerakittesting, PID: 21173
  java.lang.RuntimeException: setParameters failed
      at android.hardware.Camera.native_setParameters(Native Method)
      at android.hardware.Camera.setParameters(Camera.java:1763)
      at com.wonderkiln.camerakit.Camera1$5.run(Camera1.java:957)
      at android.os.Handler.handleCallback(Handler.java:733)
      at android.os.Handler.dispatchMessage(Handler.java:95)
      at android.os.Looper.loop(Looper.java:136)
      at android.app.ActivityThread.main(ActivityThread.java:5590)
      at java.lang.reflect.Method.invokeNative(Native Method)
      at java.lang.reflect.Method.invoke(Method.java:515)
      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1268)
      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1084)
      at dalvik.system.NativeStart.main(Native Method)
12-04 16:04:30.927 21173-21173/com.honeybeeapp.camerakittesting I/Process: Sending signal. PID: 21173 SIG: 9
```

FrantisekGazo, 2018-02-23:
I can see the same error in `v0.13.0`
```
java.lang.RuntimeException: setParameters failed
    at android.hardware.Camera.native_setParameters(Native Method)
    at android.hardware.Camera.setParameters(Camera.java:1945)
    at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:804)
    at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:779)
    at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:671)
    at com.wonderkiln.camerakit.Camera1.start(Camera1.java:124)
    at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:266)
    at android.os.Handler.handleCallback(Handler.java:790)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loop(Looper.java:164)
    at android.os.HandlerThread.run(HandlerThread.java:65)
```

austinkettner, 2018-02-23:
Can you please take a look at https://github.com/CameraKit/camerakit-android/pull/318 and comment if these issues exist within our 1.0.0 release?

emersoncloud, 2018-05-18:
Hey @Diederikjh I cloned and ran the camerakit-1.0.0 branch on my device, and Essential PH-1 and did not run into any exceptions. What device are you using and did you do a particular action to cause that exception? Thanks!