CameraKit/camerakit-android #160: Add stopPreview

guichico, 2017-08-30:
Please add a stop and a startPreview to project

akvashi, 2018-02-08:
Why was this closed? Doesnt seem to have a reason for this?

austinkettner, 2018-02-08:
This was closed as it was a stale issue, we hope to include this in our v1.0.0, reOpened as marked as a feature/enhancement

emersoncloud, 2018-08-08:
@guichico Is there a specific use case you need a startPreview and stopPreview methods for? They are currently only called internally from the `CameraKitView.onResume()` and `CamearKitView.onPause()` methods. Let me know if you need more functionality than this. 

Closing the issue but will reopen if there is still a need for this feature! =)

closed 2018-08-08 by emersoncloud

emersoncloud, 2018-08-08:
I was mistaken, we do have the ability to start and stop the preview through the `getPreviewListener()` method. The structure may change in upcoming releases but we will let you know if it does!