CameraKit/camerakit-android #161: NullPointerException on Size.getWidth()

joaogalli, 2017-08-30:
It happened on Moto Z Play with Android 7.1.1

`08-30 16:04:43.952 18192-18192/media.fppedu.fppapp E/AndroidRuntime: FATAL EXCEPTION: main
                                                                     Process: media.fppedu.fppapp, PID: 18192
                                                                     java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.Size.getWidth()' on a null object reference
                                                                         at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:358)
                                                                         at com.flurgle.camerakit.Camera1.access$200(Camera1.java:29)
                                                                         at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:71)
                                                                         at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
                                                                         at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:84)
                                                                         at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:28)
                                                                         at android.view.TextureView.getHardwareLayer(TextureView.java:390)
                                                                         at android.view.TextureView.draw(TextureView.java:339)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:16183)
                                                                         at android.view.View.draw(View.java:16967)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3731)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3517)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:16178)
                                                                         at android.view.View.draw(View.java:16967)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3731)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3517)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:16178)
                                                                         at android.view.View.draw(View.java:16967)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3731)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3517)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:16178)
                                                                         at android.view.View.draw(View.java:16967)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3731)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3517)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:16178)
                                                                         at android.view.View.draw(View.java:16967)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3731)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3517)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:16178)
                                                                         at android.view.View.draw(View.java:16967)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3731)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3517)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:16178)
                                                                         at android.view.View.draw(View.java:16967)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3731)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3517)
                                                                         at android.view.View.draw(View.java:17204)
                                                                         at com.android.internal.policy.DecorView.draw(DecorView.java:753)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:16183)
                                                                         at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:648)
                                                                         at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:654)
                                                                         at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:762)
                                                                         at android.view.ViewRootImpl.draw(ViewRootImpl.java:2818)
                                                                         at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2626)
                                                                         at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2233)
                                                                         at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1272)
                                                                         at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6408)
                                                                         at android.view.Choreographer$CallbackRecord.run(Choreographer.java:874)
                                                                         at android.view.Choreographer.doCallbacks(Choreographer.java:686)
                                                                         at android.view.Choreographer.doFrame(Choreographer.java:621)
                                                                         at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:860)
                                                                         at android.os.Handler.handleCallback(Handler.java:751)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                         at android.os.Looper.loop(Looper.java:154)
                                                                         at android.app.ActivityThread.main(ActivityThread.java:6165)
                                                                         at java.lang.reflect.Method.invoke(Native Method)
                                                                         at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
                                                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778)
`

tanis2000, 2017-09-05:
I'm having a similar issue but it happens only from time to time. It looks like some timing problem or something like that tbh.

Here's the full exception

```
09-05 10:39:32.417 25473-25473/com.kedrion.pidgenius.debug E/AndroidRuntime: FATAL EXCEPTION: main
                                                                             Process: com.kedrion.pidgenius.debug, PID: 25473
                                                                             java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.Size.getWidth()' on a null object reference
                                                                                 at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:358)
                                                                                 at com.flurgle.camerakit.Camera1.access$200(Camera1.java:29)
                                                                                 at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:71)
                                                                                 at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
                                                                                 at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:84)
                                                                                 at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:28)
                                                                                 at android.view.TextureView.getHardwareLayer(TextureView.java:390)
                                                                                 at android.view.TextureView.draw(TextureView.java:339)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16167)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
[... cut ...]
                                                                                 at android.app.ActivityThread.main(ActivityThread.java:6119)
                                                                                 at java.lang.reflect.Method.invoke(Native Method)
09-05 10:39:32.418 25473-25473/com.kedrion.pidgenius.debug E/AndroidRuntime:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
                                                                                 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
```

tanis2000, 2017-09-05:
This is still relevant for version 0.9.18

```
09-05 11:06:51.449 22425-22425/com.kedrion.pidgenius.debug E/AndroidRuntime: FATAL EXCEPTION: main
                                                                             Process: com.kedrion.pidgenius.debug, PID: 22425
                                                                             java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.Size.getWidth()' on a null object reference
                                                                                 at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:372)
                                                                                 at com.flurgle.camerakit.Camera1.access$200(Camera1.java:30)
                                                                                 at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:75)
                                                                                 at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
                                                                                 at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:80)
                                                                                 at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:26)
                                                                                 at android.view.TextureView.getHardwareLayer(TextureView.java:390)
                                                                                 at android.view.TextureView.draw(TextureView.java:339)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16167)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
                                                                                 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3513)
                                                                                 at android.view.View.updateDisplayListIfDirty(View.java:16162)
                                                                                 at android.view.View.draw(View.java:16951)
                                                                                 at android.view.ViewGroup.drawChild(ViewGroup.java:3727)
[... cut ...]
                                                                                 at android.app.ActivityThread.main(ActivityThread.java:6119)
                                                                                 at java.lang.reflect.Method.invoke(Native Method)
09-05 11:06:51.449 22425-22425/com.kedrion.pidgenius.debug E/AndroidRuntime:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
                                                                                 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
```

tanis2000, 2017-09-05:
For the time being I created a Runnable that I delay by 1000ms before running `camera.Start()`. It works but it definitely looks like a synchronization issue.

bootstraponline, 2017-10-16:
I'm seeing a similar crash.

```java
java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.Size.getWidth()' on a null object reference
FATAL EXCEPTION: CameraViewWorker
Process: com.instructure.teacher, PID: 14346
java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.Size.getWidth()' on a null object reference
    at com.flurgle.camerakit.Camera1.initResolutions(Camera1.java:384)
    at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:485)
    at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:427)
    at com.flurgle.camerakit.Camera1.start(Camera1.java:135)
    at com.flurgle.camerakit.CameraView$3.run(CameraView.java:305)
    at android.os.Handler.handleCallback(Handler.java:751)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:154)
    at android.os.HandlerThread.run(HandlerThread.java:61)
```

austinkettner, 2017-10-19:
Should be resolved in 0.11.2, please comment if you still experience this!

@joaogalli @bootstraponline @tanis2000 @aselby

closed 2017-10-19 by austinkettner