CameraKit/camerakit-android #164: Crop works for photos but not video

Mack4545, 2017-09-04:
When I invoke 

`cameraView.setCropOutput(true);`

the picture is cropped perfectly to mirror that of my cameraView's heigh. 

The video however is not. Anyone know how I can solve this?

UNCVR, 2017-09-25:
Video cropping is not supported, you have to crop the video manually

taosimple, 2017-11-01:
Should not !, recording camera also should support custom aspect ratio, and the strange thing is : in CameraKit project have function `getVideoResolution()` but not invoked, the function below : 
'''
@Override
    Size getVideoResolution() {
        Log.e(TAG, "getVideoResolution: ");
        if (mVideoSize == null && mCameraParameters != null) {
            if (mCameraParameters.getSupportedVideoSizes() == null) {
                mVideoSize = getCaptureResolution();
                return mVideoSize;
            }

            TreeSet<Size> sizes = new TreeSet<>();
            for (Camera.Size size : mCameraParameters.getSupportedVideoSizes()) {
                Log.e(TAG, "support video size : " + size.width + "/" + size.height);
                sizes.add(new Size(size.width, size.height));
            }

            TreeSet<AspectRatio> aspectRatios = findCommonAspectRatios(
                    mCameraParameters.getSupportedPreviewSizes(),
                    mCameraParameters.getSupportedVideoSizes()
            );
            AspectRatio targetRatio = aspectRatios.size() > 0 ? aspectRatios.last() : null;

            Iterator<Size> descendingSizes = sizes.descendingIterator();
            Size size;
            while (descendingSizes.hasNext() && mVideoSize == null) {
                size = descendingSizes.next();
                if (targetRatio == null || targetRatio.matches(size)) {
                    mVideoSize = size;
                    break;
                }
            }
        }
        Log.e(TAG, "return video size : " + mVideoSize.getWidth() + "/" + mVideoSize.getHeight());
        return mVideoSize;
    }
'''

closed 2018-02-03 by austinkettner

iad24, 2018-08-25:
cameraView.setCropOutput(true) does not work for me. My camera view's height is set to only half the screen of my device. but the output image's height is the whole screen height.