CameraKit/camerakit-android #168: The following classes could not be instantiated:

ErkinAkbudak, 2017-09-13:
Hi. I'm trying capturing images. I added "compile 'com.flurgle:camerakit:0.9.17'" in my gradle file and i sync gradle. Here is xml file:
```
<com.flurgle.camerakit.CameraView
                android:id="@+id/camera"
                android:layout_width="match_parent"
                android:layout_height="wrap_content"
                android:adjustViewBounds="true" />
```
I get error:
```
Failed to instantiante one or more classes
The following classes could not be instantiated
            -com.flurgle.camerakit.CameraView
```
And i am trying add this code on MainActivity
```
@Override
    protected void onResume() {
        super.onResume();
        cameraView.start();
    }

    @Override
    protected void onPause() {
        cameraView.stop();
        super.onPause();
    }
```
Error:
Cannot resolve sysmbol 'cameraView'
I did not forget "import com.flurgle.camerakit.CameraView;"
Api version: 26 OS: Ubuntu
Thanks in advance.

closed 2017-09-14 by ErkinAkbudak

jdcaperon, 2018-08-28:
Was there any advice on how this issue was resolved?

emersoncloud, 2018-09-04:
Hi @jdcaperon what issue are you running into? This post referenced a much older version of CameraKit. We're currently working on the 1.0.0 release, and you can see some basic implementation details at #318. I'll be happy to answer any questions you might have.

jdcaperon, 2018-09-08:
Thanks, my issue is referenced #409.