CameraKit/camerakit-android #176: could you add a onFrameData callback in CameraListener?

leaf918, 2017-09-20:
the frame data is important for realtime frame data processing.
thanks.

MrOnyszko, 2017-09-23:
I agree with that idea :)

dwillmc, 2017-11-01:
This is in the works now and coming soon. I want to do a bit on top of just passing the frame though. I've been adding a lot of NDK stuff lately to speed up processing and I'd like to structure that to be usable by CameraKit users. Especially with the frame callbacks where performance is incredibly important.

What sorts of things do you need to do with the frames?

sksar, 2017-11-01:
@dwillmc 
Possibly a support for a function where we could specify a frame processor function that gets each frame, and also the processed frame (maybe like color enhanced, filter effects) is returned to the camera View.....

morejump, 2017-11-10:
in my case, i want to  get each frame to use for JNI with opencv c++, and return preview!

leaf918, 2017-11-10:
@dwillmc In my case ,I have a lot of neural networks for real-time image processing，such as AR,robot vision

xymelon, 2017-11-13:
@dwillmc In my case, I need frame to use for real-time OCR.

monatis, 2017-11-17:
@dwillmc In my case, I'm trying to detect motion (i.e., page turns) and take a photo for an automatic scan app.

benjaminbarbe, 2017-12-12:
Do you have any update on this?

austinkettner, 2017-12-12:
@xymelon @monatis @benjaminbarbe Yes we plan to have this included in 1.0.0 release aimed for Jan 1, 2018. We hope to open a PR and Snapshot builds for this release, this week so you can comment, test, etc!

@dwillmc May have further input too

browningweb, 2018-02-02:
Do you have any update on this...? @dwillmc

austinkettner, 2018-02-02:
We are hoping to issue a 1.0.0 PR soon for testing and a lot of this is included. @browningweb

closed 2018-02-03 by austinkettner

edwardotis, 2018-04-20:
@austinkettner Can you post a link here to the PR and/or feature docs? This github issue gets high SEO for the topic.

eviansplash, 2018-07-21:
BUMP......? should we switch camera package from camerakit...?

austinkettner, 2018-07-21:
@eviansplash @kunaldawn @edwardotis Here is the 1.0.0 PR: https://github.com/CameraKit/camerakit-android/pull/318 give it a try and let us know a bit more about your use case? We are adding in additional calls ahead of merging and can specifically address this.