CameraKit/camerakit-android #178: Camera1.java line 333 com.flurgle.camerakit.Camera1.initResolutions NPE

AndrewGable, 2017-09-26:
Hey there, we are getting this crash primarily on Samsung devices (Galaxy S7, Galaxy S7 Edge, Note 5) 

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.Camera$Parameters.getSupportedPreviewSizes()' on a null object reference
       at com.flurgle.camerakit.Camera1.initResolutions(Camera1.java:333)
       at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:460)
       at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:430)
       at com.flurgle.camerakit.Camera1.access$200(Camera1.java:35)
       at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:88)
       at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
       at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:80)
       at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:26)
       at android.view.TextureView.getHardwareLayer(TextureView.java:373)
       at android.view.View.updateDisplayListIfDirty(View.java:15358)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.draw(View.java:16447)
       at android.view.View.updateDisplayListIfDirty(View.java:15388)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.draw(View.java:16447)
       at android.view.View.updateDisplayListIfDirty(View.java:15388)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.draw(View.java:16447)
       at android.view.View.updateDisplayListIfDirty(View.java:15388)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.updateDisplayListIfDirty(View.java:15380)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.updateDisplayListIfDirty(View.java:15380)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.updateDisplayListIfDirty(View.java:15380)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.updateDisplayListIfDirty(View.java:15380)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.updateDisplayListIfDirty(View.java:15380)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at android.view.View.updateDisplayListIfDirty(View.java:15380)
       at android.view.View.draw(View.java:16189)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3735)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3525)
       at com.android.internal.policy.PhoneWindow$DecorView.dispatchDraw(PhoneWindow.java:2754)
       at android.view.View.draw(View.java:16447)
       at com.android.internal.policy.PhoneWindow$DecorView.draw(PhoneWindow.java:2740)
       at android.view.View.updateDisplayListIfDirty(View.java:15388)
       at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:286)
       at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:292)
       at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:327)
       at android.view.ViewRootImpl.draw(ViewRootImpl.java:3020)
       at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2824)
       at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2436)
       at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1321)
       at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6708)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:894)
       at android.view.Choreographer.doCallbacks(Choreographer.java:696)
       at android.view.Choreographer.doFrame(Choreographer.java:631)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:880)
       at android.os.Handler.handleCallback(Handler.java:815)
       at android.os.Handler.dispatchMessage(Handler.java:104)
       at android.os.Looper.loop(Looper.java:207)
       at android.app.ActivityThread.main(ActivityThread.java:5730)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:789)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:679)
```

AndrewGable, 2017-09-26:
Possibly related discussion in https://github.com/gogopop/CameraKit-Android/issues/136

alexzk1, 2017-09-29:
https://github.com/alexzk1/CameraKit-Android

all that crashes are coming from 1 source ... some guy was paying 500$ on upwork, but i was not hired ..any way, free must be free... i waited 3 weeks, take it - try, see if it helps, 1 file there changed

austinkettner, 2017-09-29:
We will be integrating a fix for this along with many other fixes shortly.

Austin L Kettner                                                                          
330.714.3513 [tel:330-714-3513] - @austinkettner [https://twitter.com/austinkettner]      
www.austinkettner.com [http://www.austinkettner.com]                                      

https://github.com/alexzk1/CameraKit-Android [https://github.com/alexzk1/CameraKit-Android]

alexzk1, 2017-09-29:
Hope you got the reason ? :) ppl are using ur lib from different threads. On multicore it is really bad without sync. Check other places which may need it. However, I should say, your instructions are clear and correct, ppl must launch ur lib in diff thread and use from there. But they don't, because ur example don't :D

austinkettner, 2017-09-29:
Yep, it's understood, thank you for following up on this @alexzk1 -- also feel free to initiate a pull request.

alexzk1, 2017-09-29:
...well, it was not paid...i don't need it, i will delete repo week later or so, just make sure you have used it. If it is usefull. I was not really reading all source. It was fast fix "proove" but seems it fixes all :D

dwillmc, 2017-10-05:
This was originally caused by me merging in some work from someones PR that had a different approach to determining preview/capture resolutions. However I found that it was setting the preview size incorrectly (as low as possible) and I didn't want to use that approach anyway (what was `initResolutions`, which is now gone). Reverted this a few days ago and I just rechecked that there aren't any other uses of `mCameraParameters` without a nullcheck, so this should be all set.

closed 2017-10-05 by dwillmc