CameraKit/camerakit-android #182: Failed to resolve: com.wonderkiln:camerakit:0.10.1

bruferrari, 2017-09-28:
Hi guys, I've notice that there was an update, but I can't update it in my gradle file, seems that the  `com.wonderkiln:camerakit:0.10.1` doesn't exists.

Xenero, 2017-09-29:
Facing same issue here.

Have Invalidated/Restarted and also cleared gradle cache

BuddahLD, 2017-09-29:
Hello!
The same issue as above - failed to resolve: com.wonderkiln:camerakit:0.10.1
Invalidate/Restart didn't help.

austinkettner, 2017-09-29:
It's going to take a bit longer for Maven to update to the new package name. Should be up and available shortly! Will update this ticket once live.

bruferrari, 2017-09-29:
I'm using it with jitpack and I can confirm that works until maven processing finish

BuddahLD, 2017-09-30:
Hello!
I'm sorry, is the issue still present, or is it my-side problem?
Thank you!

BuddahLD, 2017-10-01:
Hello!
The issue is still reproducible. Sorry, is this my-side problem?
Thank you!

Xenero, 2017-10-01:
Still happening here also

austinkettner, 2017-10-01:
Hey guys. We expect this release to be ‘live’ on Monday.

Austin L Kettner                                                                          
330.714.3513 [tel:330-714-3513] - @austinkettner [https://twitter.com/austinkettner]      
www.austinkettner.com [http://www.austinkettner.com]

hazelement, 2017-10-02:
Are you guys still experiencing the same issue?

kevinstier, 2017-10-02:
Yes, I do

austinkettner, 2017-10-02:
Still waiting on Bintray here. Apologies, but it's a bit out of our control, I will post back in here as soon as it's live!

hazelement, 2017-10-02:
No worries, we understand. For the time being, I'm just adding the source code to my project to get it going. Hope it comes out soon.

austinkettner, 2017-10-03:
Hey guys, this seems to have been published and is available. Take a look!

closed 2017-10-03 by austinkettner

BuddahLD, 2017-10-03:
Hey,
Yep, it's successfully compiled into the project (:
Thank you!

hazelement, 2017-10-04:
same here. Thanks guys!