CameraKit/camerakit-android #183: startPreview Failure

rencsaridogan, 2017-10-01:
Hi,

startPreview fails at the below point with the below configuration and system state:

- OS Version: 7.0
- Device: Galaxy Note5
- RAM Free: 39.1%
- Disk Free: 9.9%
```
#0. Crashed: Thread-24: 0 0 0x0000000000000000
       at android.hardware.Camera.startPreview(Camera.java)
       at com.flurgle.camerakit.Camera1.start(Camera1.java:91)
       at com.flurgle.camerakit.CameraView$3.run(CameraView.java:216)
       at java.lang.Thread.run(Thread.java:762)

Fatal Exception: java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera.startPreview(Camera.java)
       at com.flurgle.camerakit.Camera1.start(Camera1.java:91)
       at com.flurgle.camerakit.CameraView$3.run(CameraView.java:216)
       at java.lang.Thread.run(Thread.java:762)
```
If anyone else had this issue before I kindly ask you to link it here.

Cheers,

Renç

rencsaridogan, 2017-10-01:
I believe this might be related to | same with #89

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc