CameraKit/camerakit-android #189: Error when take phot and onBackPressed

Paul75, 2017-10-07:
Hello,

I have a problem : when I take photo and do a onBackPressed, I've got :
`FAILED BINDER TRANSACTION !!!`

Thanks for help

Paul75, 2018-02-03:
Thanks for close without help .....

austinkettner, 2018-02-03:
@Paul75 We are closing stale tickets is all, we have issued a lot of versions since October, the latest being 0.13.x and a lot of these are resolved, but it's very hard to go through all tickets and comment directly.

If you test the latest version of CameraKit and still have this issue, comment and I will reOpen and attach to the 1.0.0 milestone!

Apologies if closing this made you feel we are ignoring your issue!

seanashmore, 2018-02-12:
I'm seeing this issue aswell if this could please be re-opened. Thank you.

austinkettner, 2018-02-12:
ReOpened for you!

seanashmore, 2018-02-12:
Ok so I found the solution to this in a very old stack overflow question (of course) https://stackoverflow.com/a/23424889. 

It's not a library issue, it's my own fault for trying to pass too much data as an intent extra. I don't know if this is what @Paul75 was doing but after removing the extra from my result intent the issue no longer occurred.

I hope this helps.

closed 2018-03-08 by austinkettner