CameraKit/camerakit-android #191: Can't create handler inside thread that has not called Looper.prepare()

selfimprW, 2017-10-08:
java.lang.RuntimeException: Can't create handler inside thread that has not called Looper.prepare()
                                                                       at android.os.Handler.<init>(Handler.java:200)
                                                                       at android.os.Handler.<init>(Handler.java:114)
                                                                       at android.widget.Toast$TN$2.<init>(Toast.java:343)
                                                                       at android.widget.Toast$TN.<init>(Toast.java:343)
                                                                       at android.widget.Toast.<init>(Toast.java:103)
                                                                       at android.widget.Toast.makeText(Toast.java:262)
                                                                       at com.example.zzapp.MainActivity$2.onPictureTaken(MainActivity.java:129)
                                                                       at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:524)
                                                                       at com.wonderkiln.camerakit.Camera1$3$1.onStillProcessed(Camera1.java:308)
                                                                       at com.wonderkiln.camerakit.ProcessStillTask.run(ProcessStillTask.java:47)
                                                                       at java.lang.Thread.run(Thread.java:760)

suyashbhatt, 2017-10-08:
This error is not related to the library, I bet you are not using Toast properly, check your implementation. Don't call it from worker thread

selfimprW, 2017-10-08:
You are right

suyashbhatt, 2017-10-08:
no problem, please close the issue if its resolved for you

closed 2017-10-08 by selfimprW

ananthramasamy, 2018-08-07:
Use this its work for me ....
runOnUiThread(new Runnable() {
                    public void run() {
                        Toast.makeText(getApplicationContext(), "Status = " + message , Toast.LENGTH_LONG).show();
                    }
                });

umarfarooqitgcuf, 2019-03-09:
> Use this its work for me ....
> runOnUiThread(new Runnable() {
> public void run() {
> Toast.makeText(getApplicationContext(), "Status = " + message , Toast.LENGTH_LONG).show();
> }
> });

thanx this works for me

vitalii-azarov, 2020-05-08:
> Use this its work for me ....
> runOnUiThread(new Runnable() {
> public void run() {
> Toast.makeText(getApplicationContext(), "Status = " + message , Toast.LENGTH_LONG).show();
> }
> });

You saved my day, thank you a lot!

gilly7, 2020-09-17:
> Use this its work for me ....
> runOnUiThread(new Runnable() {
> public void run() {
> Toast.makeText(getApplicationContext(), "Status = " + message , Toast.LENGTH_LONG).show();
> }
> });