CameraKit/camerakit-android #193: Error latest version: Only the original thread that ...

greywolfhalil, 2017-10-09:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](https://.github.com/docs/contributing.html)?

Yes

### Environment

Android Studio 2.3.3
LG G4 Android 6.0 API 23

### Steps to Reproduce

1. Setup build.gradle (Version 0.11.0)
2. cameraView.start() inside onResume
3. Error,

> E/UncaughtException: android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
                                                                            at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:6664)
                                                                            at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:925)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.support.constraint.ConstraintLayout.requestLayout(ConstraintLayout.java:1959)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.View.requestLayout(View.java:18768)
                                                                            at android.view.SurfaceView$4.setFixedSize(SurfaceView.java:759)
                                                                            at com.wonderkiln.camerakit.SurfaceViewPreview.setPreviewParameters(SurfaceViewPreview.java:84)
                                                                            at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:584)
                                                                            at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:570)
                                                                            at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:475)
                                                                            at com.wonderkiln.camerakit.Camera1.start(Camera1.java:113)
                                                                            at com.wonderkiln.camerakit.CameraView$3.run(CameraView.java:313)
                                                                            at android.os.Handler.handleCallback(Handler.java:739)
                                                                            at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                            at android.os.Looper.loop(Looper.java:148)
                                                                            at android.os.HandlerThread.run(HandlerThread.java:61)

> E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                     Process: io.greywolf.soruapp, PID: 24014
                                                                     android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
                                                                         at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:6664)
                                                                         at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:925)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.support.constraint.ConstraintLayout.requestLayout(ConstraintLayout.java:1959)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.View.requestLayout(View.java:18768)
                                                                         at android.view.SurfaceView$4.setFixedSize(SurfaceView.java:759)
                                                                         at com.wonderkiln.camerakit.SurfaceViewPreview.setPreviewParameters(SurfaceViewPreview.java:84)
                                                                         at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:584)
                                                                         at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:570)
                                                                         at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:475)
                                                                         at com.wonderkiln.camerakit.Camera1.start(Camera1.java:113)
                                                                         at com.wonderkiln.camerakit.CameraView$3.run(CameraView.java:313)
                                                                         at android.os.Handler.handleCallback(Handler.java:739)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                         at android.os.Looper.loop(Looper.java:148)
                                                                         at android.os.HandlerThread.run(HandlerThread.java:61)
4. When I use version 0.10.1, its working

### Expected Behavior

### Actual Behavior

### Reproducible Demo

### My layout
```
...
<FrameLayout
     android:id="@+id/layout"
     android:layout_width="0dp"
     android:layout_height="0dp"
     android:layout_marginTop="12dp"
     app:layout_constraintBottom_toBottomOf="parent"
     app:layout_constraintLeft_toLeftOf="parent"
     app:layout_constraintRight_toRightOf="parent"
     app:layout_constraintTop_toBottomOf="@id/tv_question">

     <com.wonderkiln.camerakit.CameraView xmlns:camerakit="http://schemas.android.com/apk/res-auto"
         android:id="@+id/cameraView"
         android:layout_width="match_parent"
         android:layout_height="match_parent"
         android:adjustViewBounds="true"
         camerakit:ckFacing="front"
         camerakit:ckFlash="off"
         camerakit:ckVideoQuality="max480p" />
</FrameLayout>
...
```

snkashis, 2017-10-09:
Yep, confirmed here too - Moto E Gen 2, API Level 22.

austinkettner, 2017-10-09:
This is resolved in 0.11.1 which is currently being released to Maven/Bintray. It is currently tagged in GH Releases: https://github.com/wonderkiln/CameraKit-Android/releases/tag/v0.11.1

closed 2017-10-09 by austinkettner