CameraKit/camerakit-android #194: get Error

PashkinAlexey, 2017-10-09:
10-10 18:47:32.101 30069-31358/lenremont.crm.ru.courier E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                          Process: lenremont.crm.ru.courier, PID: 30069
                                                                          android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
                                                                              at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:7011)
                                                                              at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:1042)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.View.requestLayout(View.java:17704)
                                                                              at android.view.SurfaceView$4.setFixedSize(SurfaceView.java:874)
                                                                              at com.wonderkiln.camerakit.SurfaceViewPreview.setPreviewParameters(SurfaceViewPreview.java:84)
                                                                              at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:584)
                                                                              at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:570)
                                                                              at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:475)
                                                                              at com.wonderkiln.camerakit.Camera1.start(Camera1.java:113)
                                                                              at com.wonderkiln.camerakit.CameraView$3.run(CameraView.java:313)
                                                                              at android.os.Handler.handleCallback(Handler.java:815)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:104)
                                                                              at android.os.Looper.loop(Looper.java:194)
                                                                              at android.os.HandlerThread.run(HandlerThread.java:61)

finchatticus, 2017-10-09:
im got same error with version 0.11.0

austinkettner, 2017-10-09:
@PashkinAlexey or @finchatticus Can you please provide reproduction steps? Or more details in general.

snkashis, 2017-10-09:
I was using implementation described here - https://github.com/commonsguy/cw-omnibus/blob/master/Camera/CameraKit/app/src/main/java/com/commonsware/android/camera/CameraActivity.java

austinkettner, 2017-10-09:
This is resolved in 0.11.1 which is currently being released to Maven/Bintray. It is currently tagged in GH Releases: https://github.com/wonderkiln/CameraKit-Android/releases/tag/v0.11.1

closed 2017-10-09 by austinkettner

Pomis, 2017-10-14:
Super inconvenient to use it now when it links to 26.1.0

austinkettner, 2017-10-14:
@Pomis Happy to explore allowing a lower version, can you open a ticket regarding this?