CameraKit/camerakit-android #199: OutOfMemory Crash during capture using back-facing camera

tamper2, 2017-10-11:
### Is this a bug report?
Yes

### Have you read the [Contributing Guidelines](https://github.com/wonderkiln/CameraKit-Android/.github/CONTRIBUTING.md)?
Yes

### Environment
Android Studio 3.0 Beta 7
Samsung Galaxy S5
com.wonderkiln:camerakit:0.11.1

### Steps to Reproduce

1. Load up a CameraView
2. Capture image using the back-facing camera
3. Application crashes with OutOfMemoryError during ExifUtil.decodeBitmapWithRotation

When using the front-facing camera there is no issue.
I think the crash is due to resolution differences between the front/back facing cameras.

### Expected Behavior

I thought the picture will be transferred to the CameraListener I've set up.

### Actual Behavior

Application crashes:

E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                  Process: com.orons.weddingapp, PID: 4186
                  java.lang.OutOfMemoryError: Failed to allocate a 63489036 byte allocation with 16777216 free bytes and 41MB until OOM
                      at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                      at android.graphics.Bitmap.nativeCreate(Native Method)
                      at android.graphics.Bitmap.createBitmap(Bitmap.java:939)
                      at android.graphics.Bitmap.createBitmap(Bitmap.java:912)
                      at android.graphics.Bitmap.createBitmap(Bitmap.java:843)
                      at com.wonderkiln.camerakit.ExifUtil.decodeBitmapWithRotation(ExifUtil.java:34)
                      at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:511)
                      at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:277)
                      at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1142)
                      at android.os.Handler.dispatchMessage(Handler.java:102)
                      at android.os.Looper.loop(Looper.java:145)
                      at android.os.HandlerThread.run(HandlerThread.java:61)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

Jameido, 2017-10-11:
I'm actually facing the same issue

# Environment

Android Studio 2.3.3
Device 1:
Samsung Galaxy S5 (G900F) 
Android 6.0.1
com.wonderkiln:camerakit:0.10.1 and com.wonderkiln:camerakit:0.11.1

Device 2:
Samsung Galaxy A5 (SM-A520F)
Android 7.0.0
com.wonderkiln:camerakit:0.10.1

Device 3:
Samsung Galaxy A5
Android 6.0.1
com.wonderkiln:camerakit:0.10.1

# Steps to Reproduce

Declare CameraView in xml
Capture image using the back-facing camera
Application crashes with OutOfMemoryError during ExifUtil.decodeBitmapWithRotation
When using the front-facing camera there is no issue.

# Expected Behavior

I thought the picture will be transferred to the CameraListener I've set up.

# Actual Behavior

Application crashes:
FATAL EXCEPTION: CameraViewWorker
                                                                         Process: it.wedigital.silcamykeys, PID: 32735
                                                                         java.lang.OutOfMemoryError: Failed to allocate a 63489036 byte allocation with 16771680 free bytes and 43MB until OOM
                                                                             at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                             at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                             at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
                                                                             at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
                                                                             at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
                                                                             at com.wonderkiln.camerakit.ExifUtil.decodeBitmapWithRotation(ExifUtil.java:34)
                                                                             at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:511)
                                                                             at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:277)
                                                                             at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
                                                                             at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                             at android.os.Looper.loop(Looper.java:158)
                                                                             at android.os.HandlerThread.run(HandlerThread.java:61)

AndrewGable, 2017-10-11:
FYI - This happens on the demo app as well.

```
10-11 15:18:44.765 23141-23162/com.wonderkiln.camerakit.demo E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                               Process: com.wonderkiln.camerakit.demo, PID: 23141
                                                                               java.lang.OutOfMemoryError: Failed to allocate a 63489036 byte allocation with 16766656 free bytes and 39MB until OOM
                                                                                   at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                                   at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                                   at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
                                                                                   at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
                                                                                   at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
                                                                                   at com.wonderkiln.camerakit.ExifUtil.decodeBitmapWithRotation(ExifUtil.java:34)
                                                                                   at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:511)
                                                                                   at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:277)
                                                                                   at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
                                                                                   at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                   at android.os.Looper.loop(Looper.java:158)
                                                                                   at android.os.HandlerThread.run(HandlerThread.java:61)
```

tartacutza, 2017-10-12:
It happened to me too, same on Samsung devices.
As a quick fix...although it isn't recommended by Google I've added
`android:largeHeap="true"`
in the application tag in the Manifest
and also handled the Error (again...hacking)

Hope you'll fix this soon though, it's not the safest.

pihariev, 2017-10-12:
I've faced the same issue with CameraKit on my project where I'm currently using this library. I connected  library as a module and made some changes to ExifUtil class like creating bitmap with `inSampleSize` equals to 2 just for reducing bitmap in-memory size. There is a method that I've added to ExifUtil class: 

```
private static Bitmap getSampledBitmap(byte[] picture) {
        BitmapFactory.Options options = new BitmapFactory.Options();
        options.inJustDecodeBounds = true;
        options.inSampleSize = SAMPLED_SIZE_DEFAULT; // equals to 2
        options.inJustDecodeBounds = false;
        return BitmapFactory.decodeByteArray(picture, 0, picture.length, options);
}
```

austinkettner, 2017-10-12:
We are working on a fix for this today and will be part of the 0.11.2 release. Will keep this ticket updated as it progresses!

dwillmc, 2017-10-12:
@pihariev Yeah that's how I've fixed it locally, will be out in a release sometime today.

austinkettner, 2017-10-12:
Fixed via https://github.com/wonderkiln/CameraKit-Android/pull/204

Going into testing via @AndrewGable and @capezzbr and will close if resolved.

AndrewGable, 2017-10-20:
Unfortunately this isn't fixed for the Samsung S5, this is the stack trace I get running the demo app on the latest `master`

```
10-20 16:48:35.195 7896-7941/com.wonderkiln.camerakit.demo W/art: Throwing OutOfMemoryError "Failed to allocate a 63489036 byte allocation with 16777216 free bytes and 40MB until OOM"
10-20 16:48:35.205 7896-7941/com.wonderkiln.camerakit.demo E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                             Process: com.wonderkiln.camerakit.demo, PID: 7896
                                                                             java.lang.OutOfMemoryError: Failed to allocate a 63489036 byte allocation with 16777216 free bytes and 40MB until OOM
                                                                                 at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                                 at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                                 at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
                                                                                 at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
                                                                                 at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
                                                                                 at com.wonderkiln.camerakit.ExifUtil.decodeBitmapWithRotation(ExifUtil.java:38)
                                                                                 at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:520)
                                                                                 at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:288)
                                                                                 at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                 at android.os.Looper.loop(Looper.java:158)
                                                                                 at android.os.HandlerThread.run(HandlerThread.java:61)
```

austinkettner, 2017-10-20:
Moved to 0.12.0 milestone @AndrewGable -- thanks for reOpening and the trace!

ritesh94, 2017-10-29:
Facing same in Moto G3 Android 6.0

Throwing OutOfMemoryError "Failed to allocate a 48688652 byte allocation with 16773168 free bytes and 21MB until OOM"

FATAL EXCEPTION: CameraViewWorker
                                                                             Process: com.example.ritesh.rkcamera, PID: 32260
                                                                             java.lang.OutOfMemoryError: Failed to allocate a 48688652 byte allocation with 16773168 free bytes and 21MB until OOM
                                                                                 at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                                 at android.graphics.BitmapRegionDecoder.nativeDecodeRegion(Native Method)
                                                                                 at android.graphics.BitmapRegionDecoder.decodeRegion(BitmapRegionDecoder.java:188)
                                                                                 at com.wonderkiln.camerakit.CenterCrop.<init>(CenterCrop.java:36)
                                                                                 at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:530)
                                                                                 at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:288)
                                                                                 at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1111)
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                 at android.os.Looper.loop(Looper.java:148)
                                                                                 at android.os.HandlerThread.run(HandlerThread.java:61)

abhisheksingapore, 2017-10-29:
Facing same issue on Samsung Galaxy J7 Duos. 

 `
E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                       
                                                                          java.lang.OutOfMemoryError: Failed to allocate a 38340876 byte allocation with 16777216 free bytes and 36MB until OOM
                                                                              at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                              at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
                                                                              at com.wonderkiln.camerakit.ExifUtil.decodeBitmapWithRotation(ExifUtil.java:38)
                                                                              at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:512)
                                                                              at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:280)
                                                                              at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                              at android.os.Looper.loop(Looper.java:148)
                                                                              at android.os.HandlerThread.run(HandlerThread.java:61)

`

abhisheksingapore, 2017-10-29:
Facing the same issue on Samsung Galaxy J7

`
E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                       
                                                                          java.lang.OutOfMemoryError: Failed to allocate a 38340876 byte allocation with 16777216 free bytes and 36MB until OOM
                                                                              at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                              at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
                                                                              at com.wonderkiln.camerakit.ExifUtil.decodeBitmapWithRotation(ExifUtil.java:38)
                                                                              at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:512)
                                                                              at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:280)
                                                                              at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                              at android.os.Looper.loop(Looper.java:148)
                                                                              at android.os.HandlerThread.run(HandlerThread.java:61)

````

abhisheksingapore, 2017-10-29:
Facing the same issue on Samsung Galaxy J7

` E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                          Process: me.veganbuddy.veganbuddy, PID: 16262
                                                                          java.lang.OutOfMemoryError: Failed to allocate a 38340876 byte allocation with 16777216 free bytes and 36MB until OOM
                                                                              at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                              at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
                                                                              at com.wonderkiln.camerakit.ExifUtil.decodeBitmapWithRotation(ExifUtil.java:38)
                                                                              at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:512)
                                                                              at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:280)
                                                                              at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                              at android.os.Looper.loop(Looper.java:148)
                                                                              at android.os.HandlerThread.run(HandlerThread.java:61)
`

abhisheksingapore, 2017-10-29:
Facing the same issue on Samsung Galaxy J7

` E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                          Process: me.buddy.buddy, PID: 16262
                                                                          java.lang.OutOfMemoryError: Failed to allocate a 38340876 byte allocation with 16777216 free bytes and 36MB until OOM
                                                                              at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                              at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
                                                                              at com.wonderkiln.camerakit.ExifUtil.decodeBitmapWithRotation(ExifUtil.java:38)
                                                                              at com.wonderkiln.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:512)
                                                                              at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:280)
                                                                              at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                              at android.os.Looper.loop(Looper.java:148)
                                                                              at android.os.HandlerThread.run(HandlerThread.java:61)
`

dwillmc, 2017-11-03:
Fixed in `0.12.0`. Update dependency to:
```
compile 'com.wonderkiln:camerakit:0.12.0'
```

closed 2017-11-03 by dwillmc

BharathKumarBachina, 2017-11-16:
Hi,
This issue still there in Samsung Galaxy Note5 (SM-N920I). 

Here is the log

                                                                            java.lang.OutOfMemoryError: Failed to allocate a 63489036 byte allocation with 16777216 free bytes and 30MB until OOM
                                                                                at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                                at android.graphics.BitmapRegionDecoder.nativeDecodeRegion(Native Method)
                                                                                at android.graphics.BitmapRegionDecoder.decodeRegion(BitmapRegionDecoder.java:285)
                                                                                at com.wonderkiln.camerakit.PostProcessor.getBitmap(PostProcessor.java:84)
                                                                                at com.wonderkiln.camerakit.PostProcessor.getJpeg(PostProcessor.java:42)
                                                                                at com.wonderkiln.camerakit.CameraView$5.imageCaptured(CameraView.java:439)
                                                                                at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:288)
                                                                                at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1115)
                                                                                at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                at android.os.Looper.loop(Looper.java:154)
                                                                                at android.os.HandlerThread.run(HandlerThread.java:61)

quangvinhvux, 2018-01-22:
hi team! Unfortunately this isn't fixed for the Samsung S6 /0.13.1 .please take a look.
AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                  Process: com.app.yeeld, PID: 16116
                  java.lang.OutOfMemoryError: Failed to allocate a 63489036 byte allocation with 16777216 free bytes and 59MB until OOM
                      at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                      at android.graphics.BitmapRegionDecoder.nativeDecodeRegion(Native Method)
                      at android.graphics.BitmapRegionDecoder.decodeRegion(BitmapRegionDecoder.java:285)
                      at com.wonderkiln.camerakit.PostProcessor.getBitmap(PostProcessor.java:86)
                      at com.wonderkiln.camerakit.PostProcessor.getJpeg(PostProcessor.java:41)
                      at com.wonderkiln.camerakit.CameraView$4.imageCaptured(CameraView.java:470)
                      at com.wonderkiln.camerakit.Camera1$5.onPictureTaken(Camera1.java:427)
                      at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
                      at android.os.Handler.dispatchMessage(Handler.java:102)
                      at android.os.Looper.loop(Looper.java:158)
                      at android.os.HandlerThread.run(HandlerThread.java:61)