CameraKit/camerakit-android #201: 0.11.1 changes aspect ratio after video recording begins on some devices

snkashis, 2017-10-11:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](https://github.com/wonderkiln/CameraKit-Android/.github/CONTRIBUTING.md)?

Yes

### Environment

BLU R1 HD, Android 6.0

### Steps to Reproduce

1.Set up basic video recording activity like https://github.com/commonsguy/cw-omnibus/blob/master/Camera/CameraKit/app/src/main/java/com/commonsware/android/camera/CameraActivity.java
2. Open activity
3. Push button to begin recording

### Expected Behavior

Aspect ratio of Video preview should not change.(and it doesnt, in 0.9.18)

### Actual Behavior

Video preview appears squeezed. Screen recording from Android Studio below.

https://streamable.com/1707p

### Reproducible Demo

https://github.com/commonsguy/cw-omnibus/tree/master/Camera/CameraKit

austinkettner, 2017-10-19:
Should be resolved in 0.11.2, please comment if you still experience this!

snkashis, 2017-11-10:
Still seems to be happening on 0.11.2 on the Alcatel.
Video link - https://streamable.com/bwf8j

austinkettner, 2017-11-10:
@snkashis Can you try with 0.12.0 or 0.13.0-Snapshot and let us know if you still see the issue?

snkashis, 2017-11-10:
Yes, still happening in those two versions.

snkashis, 2017-12-04:
Continues to occur in 0.12.2 on my BLU R1 as well.

dwillmc, 2017-12-05:
I've added an attribute for now before a more permanent fix will come in 1.0.0 at the end of 2017. Add `ckLockVideoAspectRatio="true"` to your cameraview and the cameraview will only choose a preview resolution that matches an available video resolution, which should solve the issue. If not re-open this and I'll dig deeper.

snkashis, 2017-12-05:
Hmm, still seem to be having issues here in 0.12.3. Take a look at following videos. 

blu r1 - https://streamable.com/z383s
alcatel 5046G- https://streamable.com/p04vu

Still fine on the moto E gen 2.

Layout looks like this
```
<?xml version="1.0" encoding="utf-8"?>
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    android:layout_width="match_parent"
    android:layout_height="match_parent">

    <com.wonderkiln.camerakit.CameraView android:id="@+id/camera"
        xmlns:camerakit="http://schemas.android.com/apk/res-auto"
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:adjustViewBounds="true"
        camerakit:ckVideoQuality="max720p"
        camerakit:ckFlash="auto"
        camerakit:ckLockVideoAspectRatio="true"
        camerakit:ckZoom="pinch" />

    <android.support.design.widget.FloatingActionButton
        android:id="@+id/fab"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_alignParentBottom="true"
        android:layout_alignParentEnd="true"
        android:layout_alignParentRight="true"
        android:layout_marginBottom="@dimen/fab_margin"
        android:layout_marginEnd="@dimen/fab_margin"
        android:layout_marginRight="@dimen/fab_margin"
        android:src="@drawable/ic_camera_black_24dp"
        app:backgroundTint="@color/colorAccent" />
</RelativeLayout>

austinkettner, 2017-12-12:
The work we are doing in 1.0.0, soon to be available via Snapshots and a PR will address this completely. Thanks for your patience!

austinkettner, 2018-03-08:
Should be resolved in #318 can you please test that out and I will reOpen this if you comment that it is still an issue.

closed 2018-03-08 by austinkettner

vikassminq, 2018-04-24:
Still i am facing this issue in 0.13.1 . its solve in v1.0.0 .

PR #271 (2017-12-05): Release v0.12.3
Resolves #201
commit 93d2805c (2017-12-05): Add ckLockVideoAspectRatio attribute for ensuring that preview aspect ratio matches a video aspect ratio

Resolves #201