CameraKit/camerakit-android #202: Attempt to invoke virtual method 'void android.hardware.Camera.setOneShotPreviewCallback(android.hardware.Camera$PreviewCallback)' on a null object reference

selfimprW, 2017-10-12:


browningweb, 2017-12-03:
I have this problem too. 
on android 8.0, 7.1.1, 6.0

`java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.setOneShotPreviewCallback(android.hardware.Camera$PreviewCallback)' on a null object reference
	at com.wonderkiln.camerakit.Camera1.captureImage(Camera1.java:313)
	at com.wonderkiln.camerakit.CameraView.captureImage(CameraView.java:431)
	at cc.ningstudio.camera.kit.CameraControls$1.onClick(CameraControls.java:122)
	at android.view.View.performClick(View.java:6289)
	at android.view.View$PerformClick.run(View.java:24800)
	at android.os.Handler.handleCallback(Handler.java:789)
	at android.os.Handler.dispatchMessage(Handler.java:98)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6809)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)`

closed 2018-02-03 by austinkettner

cuilongzhe, 2018-02-05:
@browningweb @austinkettner 
Can you update what happened here?
Why is it closed without any info?

austinkettner, 2018-02-05:
@cuilongzhe This was closed mainly as there is no recent reported issues on this using the latest versions of CameraKit. Additionally we don't have any reproduction or more detailed bug information. If you want to open a new ticket following the Issue Guidelines we will attach it to the 1.0.0 release which is coming in the next few days and ensure it's resolved prior to merging 1.0.0 in.

Hope that makes sense! =)

cuilongzhe, 2018-02-05:
@austinkettner 
The problem is that it was inevitable for me to downgrade from 0.13.1 to 0.13.0 because of #288.
If 1.0.0 is going to fix #288 as well, then I will be happy to wait 1.0.0 in which both issues get fixed hopefully.

austinkettner, 2018-02-05:
Ah makes sense, yes #288 will be resolved in 1.0.0

cuilongzhe, 2018-02-22:
@austinkettner Any update on when 1.0.0 will come out?

austinkettner, 2018-02-22:
You can start initial testing here: https://github.com/CameraKit/camerakit-android/pull/318 =)

cuilongzhe, 2018-03-22:
@austinkettner Just following up about what's going on with #318 
We have some prod users reporting this crash.
Can you please give me some ETA on this issue? Thanks in advance.

austinkettner, 2018-03-22:
@cuilongzhe We are working on pushing an update to the 1.0.0 branch/pr in the next 24hrs. It will have near full functional coverage of previous 0.13.x so it should be a good go to solution at that point. We intend to issue the full 1.0.0 by end of month as a tagged and distributed release.

thirtyyuan, 2020-01-19:
I have to use video function, so I hope "video support" come as faster as possible. Then, I can upgrade to `1.0.0`. Thx!