CameraKit/camerakit-android #212: setParameters() fails on HTC devices, causes a crash

rencsaridogan, 2017-10-19:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](https://github.com/wonderkiln/CameraKit-Android/.github/CONTRIBUTING.md)?

Yes

### Environment

Brand: HTC
Device: Various Devices
OS: Marshmallow (6.0)

### Steps to Reproduce

1. App opens
2. App redirects to the Activity where CameraKit is used
3. App crashes

### Expected Behavior

App to not crash abnormally like this.

### Actual Behavior

App crashes.

### Reproducible Demo

The problem does occur due something caused by CameraKit which I'm not sure of. Due to this, it should be investigated to produce a demo by the team of CameraKit.

### Stacktrace

```
Fatal Exception: java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParametersSubCam(Camera.java:2553)
       at android.hardware.Camera.startPreview(Camera.java:1081)
       at com.wonderkiln.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:98)
       at com.wonderkiln.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:38)
       at com.wonderkiln.camerakit.SurfaceViewPreview$1.surfaceChanged(SurfaceViewPreview.java:39)
       at android.view.SurfaceView.updateWindow(SurfaceView.java:593)
       at android.view.SurfaceView.setFrame(SurfaceView.java:301)
       at android.view.View.layout(View.java:16721)
       at com.wonderkiln.camerakit.SurfaceViewContainer.layoutChild(SurfaceViewContainer.java:52)
       at com.wonderkiln.camerakit.SurfaceViewContainer.access$000(SurfaceViewContainer.java:10)
       at com.wonderkiln.camerakit.SurfaceViewContainer$1.run(SurfaceViewContainer.java:65)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:168)
       at android.app.ActivityThread.main(ActivityThread.java:5845)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:797)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:687)
```

UmarBhutta, 2018-01-04:
It is still crashing on HTC One_M8 (Marshmallow 6.0)

jlmcdonnell, 2018-01-30:
I am seeing this on the Galaxy S6 on 7.0