CameraKit/camerakit-android #214: java.lang.RuntimeException: java.io.IOException: setPreviewTexture failed

austinkettner, 2017-10-19:
```
java.lang.RuntimeException: java.io.IOException: setPreviewTexture failed
   at com.wonderkiln.camerakit.Camera1.setupPreview(Camera1.java:507)
   at com.wonderkiln.camerakit.Camera1.access$200(Camera1.java:34)
   at com.wonderkiln.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:98)
   at com.wonderkiln.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:38)
   at com.wonderkiln.camerakit.SurfaceViewPreview$1.surfaceChanged(SurfaceViewPreview.java:39)
   at android.view.SurfaceView.updateWindow(SurfaceView.java:664)
   at android.view.SurfaceView$4.setFormat(SurfaceView.java:981)
   at com.wonderkiln.camerakit.SurfaceViewPreview$2.run(SurfaceViewPreview.java:88)
   at android.os.Handler.handleCallback(Handler.java:751)
   at android.os.Handler.dispatchMessage(Handler.java:95)
   at android.os.Looper.loop(Looper.java:154)
   at android.app.ActivityThread.main(ActivityThread.java:6688)
   at java.lang.reflect.Method.invoke(Native Method)
   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1468)
   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1358)
Caused by: java.io.IOException: setPreviewTexture failed
   at android.hardware.Camera.setPreviewSurface(Native Method)
   at android.hardware.Camera.setPreviewDisplay(Camera.java:652)
   at com.wonderkiln.camerakit.Camera1.setupPreview(Camera1.java:505)
   at com.wonderkiln.camerakit.Camera1.access$200(Camera1.java:34) 
   at com.wonderkiln.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:98) 
   at com.wonderkiln.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:38) 
   at com.wonderkiln.camerakit.SurfaceViewPreview$1.surfaceChanged(SurfaceViewPreview.java:39) 
   at android.view.SurfaceView.updateWindow(SurfaceView.java:664) 
   at android.view.SurfaceView$4.setFormat(SurfaceView.java:981) 
   at com.wonderkiln.camerakit.SurfaceViewPreview$2.run(SurfaceViewPreview.java:88) 
   at android.os.Handler.handleCallback(Handler.java:751) 
   at android.os.Handler.dispatchMessage(Handler.java:95) 
   at android.os.Looper.loop(Looper.java:154) 
   at android.app.ActivityThread.main(ActivityThread.java:6688) 
   at java.lang.reflect.Method.invoke(Native Method) 
   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1468) 
   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1358) 
```

AndrewGable, 2017-10-20:
I believe this was fixed by eb53a71bd6edacdf4d8871b35dd71bc5141a0ab5

closed 2017-11-01 by dwillmc

raigyoumei, 2018-05-31:
used version [implementation 'com.wonderkiln:camerakit:0.13.1']
------------------------------------------
Caused by java.io.IOException setPreviewTexture failed
android.hardware.Camera.setPreviewSurface (Camera.java)
android.hardware.Camera.setPreviewDisplay (Camera.java:827)
arrow_right
com.wonderkiln.camerakit.Camera1.setupPreview (Camera1.java:697)
com.wonderkiln.camerakit.Camera1.access$200 (Camera1.java:37)
com.wonderkiln.camerakit.Camera1$1.onSurfaceChanged (Camera1.java:106)
com.wonderkiln.camerakit.PreviewImpl.dispatchSurfaceChanged (PreviewImpl.java:38)
com.wonderkiln.camerakit.SurfaceViewPreview$2.surfaceChanged (SurfaceViewPreview.java:47)
android.view.SurfaceView.updateWindow (SurfaceView.java:608)
android.view.SurfaceView$3.onPreDraw (SurfaceView.java:179)
android.view.ViewTreeObserver.dispatchOnPreDraw (ViewTreeObserver.java:944)
android.view.ViewRootImpl.performTraversals (ViewRootImpl.java:2055)
android.view.ViewRootImpl.doTraversal (ViewRootImpl.java:1107)
android.view.ViewRootImpl$TraversalRunnable.run (ViewRootImpl.java:6013)
android.view.Choreographer$CallbackRecord.run (Choreographer.java:858)
android.view.Choreographer.doCallbacks (Choreographer.java:670)
android.view.Choreographer.doFrame (Choreographer.java:606)
android.view.Choreographer$FrameDisplayEventReceiver.run (Choreographer.java:844)
android.os.Handler.handleCallback (Handler.java:739)
android.os.Handler.dispatchMessage (Handler.java:95)
android.os.Looper.loop (Looper.java:148)
android.app.ActivityThread.main (ActivityThread.java:5554)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:746)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:636)

raigyoumei, 2018-05-31:
used version [implementation 'com.wonderkiln:camerakit:0.13.1']
-------------------------------------------------------------------------
Fatal Exception: java.lang.RuntimeException java.io.IOException: setPreviewTexture failed
com.wonderkiln.camerakit.Camera1.setupPreview (Camera1.java:699)
com.wonderkiln.camerakit.Camera1.access$200 (Camera1.java:37)
com.wonderkiln.camerakit.Camera1$1.onSurfaceChanged (Camera1.java:106)
com.wonderkiln.camerakit.PreviewImpl.dispatchSurfaceChanged (PreviewImpl.java:38)
com.wonderkiln.camerakit.SurfaceViewPreview$2.surfaceChanged (SurfaceViewPreview.java:47)
android.view.SurfaceView.updateWindow (SurfaceView.java:608)
android.view.SurfaceView$3.onPreDraw (SurfaceView.java:179)
android.view.ViewTreeObserver.dispatchOnPreDraw (ViewTreeObserver.java:944)
android.view.ViewRootImpl.performTraversals (ViewRootImpl.java:2055)
android.view.ViewRootImpl.doTraversal (ViewRootImpl.java:1107)
android.view.ViewRootImpl$TraversalRunnable.run (ViewRootImpl.java:6013)
android.view.Choreographer$CallbackRecord.run (Choreographer.java:858)
android.view.Choreographer.doCallbacks (Choreographer.java:670)
android.view.Choreographer.doFrame (Choreographer.java:606)
android.view.Choreographer$FrameDisplayEventReceiver.run (Choreographer.java:844)
android.os.Handler.handleCallback (Handler.java:739)
android.os.Handler.dispatchMessage (Handler.java:95)
android.os.Looper.loop (Looper.java:148)
android.app.ActivityThread.main (ActivityThread.java:5554)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:746)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:636)

raigyoumei, 2018-05-31:
plsease help me