CameraKit/camerakit-android #215: "Only the original thread..." on camera.start()

Kiruel, 2017-10-20:
Hi community !

I have a crash when I start the camera. Instant crash. On simulation Nexus5xAPI26. And i test with a OnePlus 2 same issue.

Log:
`E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                  Process: com.monaize.monaize_app.debug, PID: 5692
                  android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
                      at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:7286)
                      at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:1155)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.support.constraint.ConstraintLayout.requestLayout(ConstraintLayout.java:1959)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.view.View.requestLayout(View.java:21922)
                      at android.view.SurfaceView$3.setFixedSize(SurfaceView.java:955)
                      at com.wonderkiln.camerakit.SurfaceViewPreview.setPreviewParameters(SurfaceViewPreview.java:84)
                      at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:584)
                      at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:570)
                      at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:475)
                      at com.wonderkiln.camerakit.Camera1.start(Camera1.java:113)
                      at com.wonderkiln.camerakit.CameraView$3.run(CameraView.java:313)
                      at android.os.Handler.handleCallback(Handler.java:789)
                      at android.os.Handler.dispatchMessage(Handler.java:98)
                      at android.os.Looper.loop(Looper.java:164)
                      at android.os.HandlerThread.run(HandlerThread.java:65)`

pawelcala, 2017-10-23:
@Kiruel Hey, you should update CameraKit version to:
`compile 'com.wonderkiln:camerakit:0.11.1'`

Kiruel, 2017-10-23:
@pawelcala I compile with this version.
Finally it's work with previous version.

austinkettner, 2017-10-24:
Awesome @Kiruel -- you can also use the latest `compile 'com.wonderkiln:camerakit:0.11.2' or `compile 'com.wonderkiln:camerakit:0.11.3-SNAPSHOT' -- Snapshot builds are not as stable as non-snapshot builds though!

closed 2017-10-24 by austinkettner