CameraKit/camerakit-android #220: A bit cropped images on output accords to no common aspect ratios.

pihariev, 2017-10-24:
### Is this a bug report?

Yes.

### Have you read the [Contributing Guidelines](https://github.com/wonderkiln/CameraKit-Android/.github/CONTRIBUTING.md)?

No, it's not available.

### Environment

Samsung S5 (5.0), HTC M9 (6.0), HTC One mini 2 (4.4.2), Huawei Nexus 6P (7.1).

### Steps to Reproduce

1. Make photo with CameraKit and check preview borders.
2. Take a look at output image and compare borders.
3. Borders aren't similar.

### Expected Behavior

Preview's and captured image's borders are similar.

### Actual Behavior

[image]
[image]

### Reproducible Demo

It can be reproduced on any device from environment list.

austinkettner, 2017-10-24:
Thanks @pihariev -- slotted this into the 0.12.0 release!

pihariev, 2017-10-24:
I've fixed this one already for my own project, you can check it [here](https://github.com/wonderkiln/CameraKit-Android/pull/221).

pihariev, 2017-10-26:
@austinkettner, this bug still reproduces on devices with nonstantard displays. Could you reopen it please? I've check it on Samsung Galaxy S8 where preview's aspect ratio is 349:180. If I have a time I'll try to find a solution. Here is some screens from S8: 

[image]
[image]

austinkettner, 2017-10-26:
Done @pihariev

pihariev, 2017-10-27:
@austinkettner, I've already fixed it, you can chek solution [here](https://github.com/wonderkiln/CameraKit-Android/pull/225).

austinkettner, 2017-10-29:
Merged on in!

closed 2017-10-29 by austinkettner

fabiangoettlicher, 2017-11-20:
I have the same problem when recording video with version 0.12.0.
The borders of the recorded video are not the same as shown in preview (they are zoomed in)

gSrikar, 2017-11-24:
Great that this issue was [resolved](https://github.com/wonderkiln/CameraKit-Android/pull/221/files#diff-0b6c3dcc9c38f7368b86e52d1365bbc6). I still have this issue?

I can see the updated code in the [Camera1.class](https://github.com/wonderkiln/CameraKit-Android/blob/37dc99b521218bb5248f98432fe2d177ce9ab7af/camerakit/src/main/api16/com/wonderkiln/camerakit/Camera1.java) (Line 698 - 701) but the dependencies (0.11.2, 0.12.1) aren't adding the updated code. I'm still seeing the old code.

aravindhkumar23, 2018-01-03:
@pihariev I am also facing the cropped output for samsung s8+. The version which i used is camerakit:0.11.2. Is it resolved ? if yes let me know the version which has the fix so that i will change to that particular version.

PR #221 (2017-10-25): Fix on cropped images without proper aspect ratio options.
Fix for [this issue](https://github.com/wonderkiln/CameraKit-Android/issues/220).

Resolve #220