CameraKit/camerakit-android #227: Camera preview stretches sometimes

developer--, 2017-10-31:
At first thanks for awesome library.

I had a problem, when app is running sometimes preview is stretched automatically, but if I capture a photo it takes correct scaled picture and than preview is correct.

Tested it on Lenovo and HTC; OS v 6.0.1

dc297, 2017-11-03:
I am also facing this issue. Tried on Android 7.1.2 on Oneplus one. It happens randomly though.

tartacutza, 2017-11-06:
If you have autorotate on it will initialize with the phone's sensor values at creation. If you take a photo or rotate the phone it will recreate with the good values and not stretched. At least this is the conclusion I've reached.

devoma, 2017-11-18:
I have the same problem. When I start to record the video is stretched and after recording it becomes correct. 
Device: LG K8.  Android 6.0. 
And I noticed something. I tried another [CameraView](https://github.com/natario1/CameraView) library. There if the video quality is set to Lowest or 480p, then it is stretched like here, and if you put other qualities, it works correctly!
Maybe this information will help to understand the problem

austinkettner, 2017-11-19:
You can expect fixes here in the next few days and they will be instantly available via SNAPSHOT builds of 0.13.0!

jamolkhon, 2017-11-27:
Any update on this? Can't find any snapshot versions.

devoma, 2017-12-13:
Have you fixed this bug?

frostblooded, 2017-12-16:
Seems like this bug occurs always if the screen orientation is portrait, but the preview looks fine when the orientation is landscape.

austinkettner, 2018-02-11:
Resolved in our 1.0.0 https://github.com/CameraKit/camerakit-android/pull/318 -- please test it out using the notes in the PR and comment in the PR if you see any further issues on any devices!

closed 2018-02-11 by austinkettner