CameraKit/camerakit-android #229: onVideoTaken is not being called

virengujariya, 2017-11-03:
### Is this a bug report?
Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

No

### Environment
Android OS 7.0
Device  Moto G4 Plus

### Steps to Reproduce
`startRecordingVideo` and then `stopRecordingVideo` but it didn't call`onVideoTaken` at all.
However, capturing image does call `onPictureTaken`

(Write your steps here:)

1. start recording 
2. stop recording
3. deug to see if returns to method `onVideoTaken`

### Expected Behavior
It should call `onVideoTaken` with a file.

(Write what you thought would happen.)

### Actual Behavior

`onVideoTaken` is not being called at all.

virengujariya, 2017-11-03:
My code is ```mCameraView.setCameraListener(new CameraListener() {
            @Override
            public void onCameraOpened() {
                super.onCameraOpened();
            }

            @Override
            public void onCameraClosed() {
                super.onCameraClosed();
            }

            @Override
            public void onPictureTaken(byte[] picture) {
                super.onPictureTaken(picture);
                // Create a bitmap
                Bitmap result = BitmapFactory.decodeByteArray(picture, 0, picture.length);
            }

            @Override
            public void onVideoTaken(File video) {
                super.onVideoTaken(video);
                Toast.makeText(getActivity(), "Wow!", Toast.LENGTH_SHORT).show();
            }
        });
        mButtonRecord.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View view) {
                if (mButtonRecord.getText().toString().equalsIgnoreCase("start")) {
                    mButtonRecord.setText("stop");
                    mCameraView.captureImage();
                } else if (mButtonRecord.getText().toString().equalsIgnoreCase("stop")) {
                    mButtonRecord.setText("start");
                    mCameraView.stopRecordingVideo();
                }
            }
        });```

scottblechman, 2017-11-05:
I can reproduce this on a Nexus 5X with 8.1 and a Nexus 6 with 7.1.1. Same results with start and stop callbacks and `onPictureTacken` callback, but not `onVideoTaken`.

Hapsidra, 2017-11-14:
Try to give permission to storage. It's help for me

scottblechman, 2017-11-14:
Can still reproduce with storage permission granted

UnknownGod, 2017-11-22:
From what I could see, it seems the issue is caused by the camera capture requiring a "cached video" while recording, but it only checks for an option to save the temp video in an external storage in the prepareMediaRecorder() method of Camera1 (Camera1.java):

```
private File getVideoFile() {
    if (!Environment.getExternalStorageState().equalsIgnoreCase(Environment.MEDIA_MOUNTED)) {
        return null;
    }

    File mediaStorageDir = new File(Environment.getExternalStorageDirectory(), "Camera");

    if (!mediaStorageDir.exists()) {
        if (!mediaStorageDir.mkdirs()) {
            return null;
        }
    }

    return new File(mediaStorageDir.getPath() + File.separator + "video.mp4");
}
```

I made a minor workaround - didn't test all cases, but if you add a Context to the constructing method of Camera1 and use it in CameraView.java file:

`mCameraImpl = new Camera1(getContext(), mEventDispatcher, mPreviewImpl);`

You can then look for the internal storage directory as-well in the getVideoFile method this way:

```
private File getVideoFile() {
    if (!Environment.getExternalStorageState().equalsIgnoreCase(Environment.MEDIA_MOUNTED)) {
        return null;
    }

    File mediaStorageDir = new File(Environment.getExternalStorageDirectory(), "Camera");
    boolean missingDir = false;

    if (!mediaStorageDir.exists()) {
        if (!mediaStorageDir.mkdirs()) {
            missingDir = true;
        }
    }

    if (missingDir)
    {
        mediaStorageDir = new File(ctx.getFilesDir(), "Camera");

        if (!mediaStorageDir.exists()) {
            if (!mediaStorageDir.mkdirs()) {
                return null;
            }
        }
    }

    return new File(mediaStorageDir.getPath() + File.separator + "video.mp4");
}
```

And then the video works.

Not sure if its the best way to do it though

caleb-allen, 2018-02-08:
I'm having this same issue. Camera1 is only looking at external storage

valterh4ck3r, 2018-03-30:
Same issue

emersoncloud, 2018-07-30:
This issue will be covered with the video implementation in v1.0.0. Expecting a beta release including videos support shortly. Closing this issue, if there are still storage concerns I will create a new ticket for the 1.0.0 release!

closed 2018-07-30 by emersoncloud