CameraKit/camerakit-android #231: Load time delayed for preview

ritesh94, 2017-11-04:
Thanks for the new Release.

### Is this a bug report?
No. Its Optimization

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?
No

### Environment

Android Studio 3.0
Moto G3
Wonderkiln  - CameraKit v0.12.0

### Steps to Reproduce

1. Capture a photo
2. Delay in returning a preview of the captured image

### Expected Behavior

As soon as the image is captured the method should return the image file for the preview immediately.

### Actual Behavior

As of now, there is a delay to see the preview of the image.

Below are the logs :

11-04 08:32:41.740 6515-6619/com.example.ritesh.rkcamera D/OpenGLRenderer: endAllActiveAnimators on 0xb86b1328 (RippleDrawable) with handle 0xb87d6a68
11-04 08:32:48.587 6515-6617/com.example.ritesh.rkcamera I/art: Starting a blocking GC Alloc
11-04 08:32:48.587 6515-6617/com.example.ritesh.rkcamera I/art: Starting a blocking GC Alloc
11-04 08:32:48.601 6515-6617/com.example.ritesh.rkcamera I/art: Alloc sticky concurrent mark sweep GC freed 6535(680KB) AllocSpace objects, 0(0B) LOS objects, 11% free, 69MB/78MB, paused 413us total 13.288ms
11-04 08:32:48.601 6515-6617/com.example.ritesh.rkcamera I/art: Starting a blocking GC Alloc
11-04 08:32:48.629 6515-6617/com.example.ritesh.rkcamera I/art: Alloc partial concurrent mark sweep GC freed 403(33KB) AllocSpace objects, 3(54MB) LOS objects, 40% free, 14MB/24MB, paused 377us total 28.141ms
11-04 08:32:49.110 6515-6617/com.example.ritesh.rkcamera W/ExifInterface: Skip the tag entry since tag number is not defined: 2
11-04 08:32:51.575 6515-9290/com.example.ritesh.rkcamera I/art: Starting a blocking GC Alloc
11-04 08:32:51.575 6515-9290/com.example.ritesh.rkcamera I/art: Starting a blocking GC Alloc
11-04 08:32:51.582 6515-9290/com.example.ritesh.rkcamera I/art: Alloc sticky concurrent mark sweep GC freed 442(45KB) AllocSpace objects, 7(652KB) LOS objects, 14% free, 65MB/76MB, paused 412us total 7.002ms
11-04 08:32:51.583 6515-9290/com.example.ritesh.rkcamera I/art: Starting a blocking GC Alloc
11-04 08:32:51.612 6515-9290/com.example.ritesh.rkcamera I/art: Alloc partial concurrent mark sweep GC freed 31(1152B) AllocSpace objects, 1(49MB) LOS objects, 39% free, 15MB/26MB, paused 350us total 29.176ms
11-04 08:32:54.000 6515-9290/com.example.ritesh.rkcamera I/System.out: ImageDetails : /storage/emulated/0/saved_images/Image-6028.jpg **--** file:///storage/emulated/0/saved_images/Image-6028.jpg
11-04 08:32:54.367 6515-6515/com.example.ritesh.rkcamera D/BitmapLoadUtils: maxBitmapSize: 1385
11-04 08:32:54.370 6515-9404/com.example.ritesh.rkcamera D/BitmapWorkerTask: Uri scheme: file
11-04 08:32:54.643 6515-6515/com.example.ritesh.rkcamera D/TransformImageView: Image size: [732:1040]

austinkettner, 2017-11-05:
Looking into it @ritesh94 Thanks for the log!

MuhammedRefaat, 2017-11-09:
is there is any progress here ?!! I believe I'm facing the same issue

MuhammedRefaat, 2017-11-09:
I had the same issue and finally found the solution for it, kindly have a lock at my post here https://github.com/wonderkiln/CameraKit-Android/issues/235

austinkettner, 2017-11-20:
Thanks for the update @MuhammedRefaat -- we will include a note in the documentation!

karandeep26, 2017-11-25:
The quality of the photo is not good in case of 'still method'. is there any way to make fast the capture in standard mode?

austinkettner, 2017-11-25:
We will be addressing this in 0.13.0 @karandeep26 and we will look at setting sane defaults =)

emersoncloud, 2018-07-30:
This issue has been resolved by v1.0.0. If you are still having issues with capture speed in 1.0.0 I will create a new ticket to track it!

closed 2018-07-30 by emersoncloud