CameraKit/camerakit-android #235: Huge Delay for captured photo callback

MuhammedRefaat, 2017-11-09:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Android Nougat on Samsung Note5

### Steps to Reproduce

Capture an Image using `cameraView.captureImage(callback)` method

(Write your steps here:)

1. initiating an activity with cameraView and capture Button.
2. press the Button to call `cameraView.captureImage(callback)` method

### Expected Behavior

an Image taken and an immediate callback received at the mentioned callback in captureImage method

### Actual Behavior

Image captured successfully, but I have to wait from 3-4 seconds to receive the callback

### Reproducible Demo

no other projects reproduce that issue.

MuhammedRefaat, 2017-11-09:
I found the issue, when setting camera preferences through resource declaration in xml, they are not applied, but when setting them programmatically it can read them and they are applied.

My issue was in capture method that should be set to `still ` not  `standard`, so all what I did is :

`  cameraView.setMethod(CameraKit.Constants.METHOD_STILL);`

And now it is working perfectly and image capture callback just received directly.

closed 2017-11-09 by MuhammedRefaat

softmarshmallow, 2019-02-18:
> I found the issue, when setting camera preferences through resource declaration in xml, they are not applied, but when setting them programmatically it can read them and they are applied.
>
> My issue [...]

This won't work any longer

subhan9090, 2020-05-11:
This won't work any longer.
Please help!!