CameraKit/camerakit-android #239: TODO title

Diederikjh, 2017-11-13:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes (mostly)

### Environment

Android studio 2.3.1 (running on Ubuntu) Build tools 27.0.1 Target SDK : 27
Camera kit version: 0.12.0

### Steps to Reproduce

1. CameraKit's containing activity locked to `portrait` orientation ( `android:screenOrientation="portrait"` in Manifest file)
2. Ensure Dev settings -> `Don't keep activities` is set.
3. Launch camera activity with device in portrait mode.
4. Turn screen off.
5. Tilt device to landscape.
6. Turn screen on. (Lock screen on Samsung SM-T705 device renders in landscape orientation).
7. Swipe on unlock screen.
8. App crashes.

Stack trace: 
```
     Process: com.honeybeeapp.camerakittesting, PID: 19731
      java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.reconnect()' on a null object reference
          at com.wonderkiln.camerakit.Camera1.setupPreview(Camera1.java:521)
          at com.wonderkiln.camerakit.Camera1.start(Camera1.java:120)
          at com.wonderkiln.camerakit.CameraView$3.run(CameraView.java:297)
          at android.os.Handler.handleCallback(Handler.java:739)
          at android.os.Handler.dispatchMessage(Handler.java:95)
          at android.os.Looper.loop(Looper.java:148)
          at android.os.HandlerThread.run(HandlerThread.java:61)
```

### Expected Behavior

Camera activity would resume as normal. It does if `Do not keep activities` is not set.

### Actual Behavior

App crashed.

### Reproducible Demo

https://github.com/Diederikjh/camerakit-crash
(Note steps to reproduce as above.)

(Our camera activity is locked to portrait mode, to keep the preview sane. Preview is skewed when device is in landscape mode.

`Don't keep activities` is a rare but [valid](https://stackoverflow.com/a/22402360/8524) case for Android, and activities must to survive in that mode.)

Danm72, 2018-01-16:
Did you come up with any work arounds for this?

Diederikjh, 2018-01-16:
Can't reproduce this problem on my Android studio 3 setup.  Closing.

closed 2018-01-16 by Diederikjh

Danm72, 2018-01-16:
Why would your editor have anything to do with this? It's either device or code related. In my case I'm seeing it on lower end tablet devices.

Diederikjh, 2018-01-16:
I know, I can't really explain it either. (Android studio did upgrade gradle on the project, but couldn't' reproduce on older version either)  

Tried to reproduce on a lower end Samsung SM-T231 running Android 4.4.2.  If you can reproduce, please reopen.