CameraKit/camerakit-android #240: Null image data on callback when running on Android 4.4 device

Diederikjh, 2017-11-14:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes, mostly

### Environment

Android studio 2.3.1 (running on Ubuntu) Build tools 27.0.1 Target SDK : 27
Camera kit version: 0.12.0
Seeing this on Android 4.4 devices. (Samsung SM-T561 and SM-T231)

### Steps to Reproduce

1. Launch linked project's app.
2. Press capture button (it calls `CameraView.captureImage()`)
3. Note the toast says image callback's JPEG data is null.

### Expected Behavior

Call back to provide non-null image data.

### Actual Behavior

Callback returned null JPEG image data.

### Reproducible Demo

https://github.com/Diederikjh/camerakit-crash/tree/android-4.4-null-image

This doesn't seem to happen on version 0.11.2 of the library.

dwillmc, 2017-11-14:
Thanks for this issue. I'll have this fix out on an official release artifact within a week, but for now you can access the fix on a snapshot artifact.

Made a PR on your reproduction repo showing how to do this: https://github.com/Diederikjh/camerakit-crash/pull/1

Diederikjh, 2017-11-14:
Thanks, I'm getting correct data back on my 4.4 device.

aravindhkumar23, 2017-11-14:
Hi @Diederikjh 

did u accessed camera listener
```
// set listener for camera
        cameraView.setCameraListener(new CameraListener() {
            @Override
            public void onPictureTaken(byte[] picture) {
               // my code goes here
            }
        });
``` 

i am getting error in when using gradle
>  compile 'com.wonderkiln:camerakit:0.12.0'
> compile group: 'com.wonderkiln', name:'camerakit', version:'0.12.1-Snapshot', changing: true

errors when listener is included,
> cannot find symbol class CameraListener
> cannot resolve method setCameraListener()

dwillmc, 2017-11-14:
@aravindhkumar23 

Remove `compile 'com.wonderkiln:camerakit:0.12.0'`, only keep the line `compile group: 'com.wonderkiln', name:'camerakit', version:'0.12.1-Snapshot', changing: true`

You've upgraded from a version older than 0.12.0, where event handling has changed a bit. Change it to:

```

        cameraView.addCameraKitListener(new CameraKitEventListenerAdapter() {
            @Override
            public void onEvent(CameraKitEvent event) {
                super.onEvent(event);
            }

            @Override
            public void onError(CameraKitError error) {
                super.onError(error);
            }

            @Override
            public void onImage(CameraKitImage image) {
                super.onImage(image);
            }

            @Override
            public void onVideo(CameraKitVideo video) {
                super.onVideo(video);
            }
        });
```

aravindhkumar23, 2017-11-14:
@dwillmc 

Thanks, it seems the [documentation](http://docs.camerakit.website/#/?id=capturing-images) is not updated.

daniestrella1, 2017-11-14:
Same behavior with Android 4.4.2 and Camera kit version 0.12.0

```
Fatal Exception: java.lang.NullPointerException
       at com.wonderkiln.camerakit.CameraKitImage.getBitmap(CameraKitImage.java:20)
```

closed 2018-02-03 by austinkettner

avaneesh-reglobe, 2018-07-04:
This is fixed or not?

avaneeshkumarmaurya, 2018-07-04:
This is fixed or not?
I am getting issue of null image in callback

emersoncloud, 2018-07-18:
Hi @avaneeshkumarmaurya we've actually moved most of our development efforts to version 1.0.0. This issue is resolved there. Check the PR #318 to get started with the latest beta. We'll have an official release shortly!