CameraKit/camerakit-android #242: Crash on 0.13.0-Snapshot

AviranAbady, 2017-11-15:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Sure

### Environment

Android 8.0 / Nexus 6P

### Steps to Reproduce

Change screen orientation a few times while camrea view is engaged, im my specific case the app is locked in portrait mode so no activity/fragment recration occurs.

### Actual Behavior

Crashed

 java.lang.NullPointerException: Attempt to invoke virtual method 'void  android.hardware.Camera.reconnect()' on a null object reference
   at com.wonderkiln.camerakit.Camera1.setupPreview(Camera1.java:622)
   at com.wonderkiln.camerakit.Camera1.start(Camera1.java:116)
   at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:287)
   at android.os.Handler.handleCallback(Handler.java:789)
   at android.os.Handler.dispatchMessage(Handler.java:98)
   at android.os.Looper.loop(Looper.java:164)
   at android.os.HandlerThread.run(HandlerThread.java:65)

AviranAbady, 2017-11-18:
You can ignore this one, I believe I was handeling the camera events wrong and maybe trying to take a shot while it was not available.

sugarmanz, 2018-01-16:
I received the same error when calling `camera.start()` after `camera.stop()`, which I assume is the same thing happening to @AviranAbady (confirm?) due to the lifecycle calls on orientation change.

emersoncloud, 2018-03-19:
@sugarmanz Are you still having an issue? If yes, please fill out the Issue Template and report back here, I will reOpen the ticket. Thanks!

closed 2018-03-19 by emersoncloud