CameraKit/camerakit-android #243: Also support Kotlin ?

burkaslarry, 2017-11-16:
### Is this a bug report?

(write your answer here)

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

(Write your answer here.)

### Environment

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

austinkettner, 2017-11-16:
Hey @burkaslarry since Kotlin is fully interop, CameraKit should work in Kotlin without issue! Do you run into issues?

closed 2017-11-20 by austinkettner

iftekhar-ahmed, 2017-11-28:
I ran into this absurd issue trynna upgrade to version 0.12.x. After a full build I get hundreds of previously nonexistent nullability errors in all my kotlin files. This happens only for this particular library (and versions > 0.11.0) when added as gradle library dependency. Would appreciate if anyone comes up with a clue or faced similar problems.

burkaslarry, 2017-11-29:
what if 1.2.0 ?

austinkettner, 2017-11-29:
@iftekhar-ahmed Can you outline what version of Kotlin and some example errors?

burkaslarry, 2017-11-29:
I have used kotlin 1.2.0  . seems ok .could you please tell me ....any examples of using captureImage with completion handler  ?

burkaslarry, 2017-11-29:
I have sen the doc , no page for mentioning this.

iftekhar-ahmed, 2017-11-29:
@austinkettner I tried kotlin **1.1.60**. Here is sample build errors -

library version - **0.12.1**

`e: /Workspace/tise-android/app/src/main/java/com/tise/tise/adapters/AbsFragmentPagerAdapter.kt: (29, 5): 'destroyItem' overrides nothing
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/adapters/AbsFragmentPagerAdapter.kt: (31, 27): Type mismatch: inferred type is ViewGroup? but ViewGroup was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (119, 68): Type mismatch: inferred type is Context? but Context was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (121, 53): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type FragmentActivity?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (137, 64): Type mismatch: inferred type is Context? but Context was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (139, 49): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type FragmentActivity?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (161, 52): Type mismatch: inferred type is Context? but Context was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (163, 37): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type FragmentActivity?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (186, 27): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type Bundle?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (187, 27): Smart cast to 'AppCompatActivity' is impossible, because 'activity' is a property that has open or custom getter
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (216, 43): Type mismatch: inferred type is Context? but Context was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (227, 5): 'onCreateView' overrides nothing
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (232, 5): 'onViewCreated' overrides nothing
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (233, 29): Type mismatch: inferred type is View? but View was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (243, 86): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type Context?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AddToUDLDialogFragment.kt: (269, 18): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type Bundle?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AttributeValuesListFragment.kt: (61, 33): Type parameter bound for T in val <T : Any> T.javaClass: Class<T>
 is not satisfied: inferred type FragmentActivity? is not a subtype of Any
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AttributeValuesListFragment.kt: (61, 43): Type inference failed: Not enough information to infer parameter T in val <T : Any!> Class<T>.simpleName: String!
Please specify it explicitly.

e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/AttributeValuesListFragment.kt: (73, 38): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type Bundle?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/BottomBarFragment.kt: (33, 51): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type Context?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/BottomBarFragment.kt: (130, 45): Type mismatch: inferred type is Context? but Context was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/CameraRollFragment.kt: (89, 5): 'onCreateView' overrides nothing
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/CameraRollFragment.kt: (92, 5): 'onViewCreated' overrides nothing
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/CameraRollFragment.kt: (93, 29): Type mismatch: inferred type is View? but View was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/CameraRollFragment.kt: (125, 36): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type FragmentActivity?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/SearchFilterFragment.kt: (227, 41): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type Context?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/SearchFilterFragment.kt: (230, 16): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type Context?
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/SearchFilterFragment.kt: (385, 75): Type mismatch: inferred type is Context? but Context was expected
[... cut ...]
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/UDLOptionsDialogFragment.kt: (56, 29): Type mismatch: inferred type is View? but View was expected
e: /Workspace/tise-android/app/src/main/java/com/tise/tise/fragments/UDLOptionsDialogFragment.kt: (57, 33): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type Bundle?

:app:compileDebugKotlin FAILED
:app:buildInfoGeneratorDebug

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugKotlin'.
> Compilation error. See log for more details
`

iftekhar-ahmed, 2018-04-24:
@froesmatheus I ended up ditching the library. Later discovered the library had nothing to do with the kotlin errors. Google on some appcompat version update, decided all getContext(), getActivity() calls are candidates for null return types and kotlin compiler just put the icing on the cake(!). That's why so many errors showed up. You gotta put the Force RTE !! or check nulls with elvis ?: here