CameraKit/camerakit-android #244: Camera is being used after Camera.release() was called

niqmk, 2017-11-17:
> FATAL EXCEPTION: CameraViewWorker
>                                                                     Process: com.helpio.ubiklan, PID: 13861
>                                                                     java.lang.RuntimeException: Camera is being used after Camera.release() was called
>                                                                         at android.hardware.Camera.native_getParameters(Native Method)
>                                                                         at android.hardware.Camera.getParameters(Camera.java:3101)
>                                                                         at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1303)
>                                                                         at android.os.Handler.dispatchMessage(Handler.java:110)
>                                                                         at android.os.Looper.loop(Looper.java:208)
>                                                                         at android.os.HandlerThread.run(HandlerThread.java:61)

```
@Override
	protected void onPause() {
		Handler handler = new Handler();
		if (cameraView != null) {
			handler.postDelayed(new Runnable() {
				@Override
				public void run() {
					cameraView.stop();
				}
			}, 700);
		}
		super.onPause();
	}
```

@Overr```
ide
	protected void onResume() {
		super.onResume();
		if (isPermitted()) {
			if (isStopped) {
				isStopped = false;
				cameraView.start();
			} else {
				Handler handler = new Handler();
				if (cameraView != null) {
					handler.postDelayed(new Runnable() {
						@Override
						public void run() {
							try {
								if (!cameraView.isStarted()) {
									cameraView.start();
								}
							} catch (Exception ex) {
								ex.printStackTrace();
								if (!extCameraOpened) {
									setExtCamera();
								}
							}
						}
					}, 700);
				}
			}
		} else {
			requestPermission();
		}
	}
```

niqmk, 2017-11-20:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

### Steps to Reproduce

(Write your steps here:)

1. The Code Above
2. Reproduce if you're using ASUS X008DA ANDROID OS 7

### Expected Behavior

Exception

### Actual Behavior

java.lang.RuntimeException: Camera is being used after Camera.release() was called

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

sagartrehan, 2017-11-21:
Facing same issue

emersoncloud, 2018-07-30:
This issue has been resolved in 1.0.0 of CameraKit. If you have any further issues releasing the camera, let us know!

closed 2018-07-30 by emersoncloud