CameraKit/camerakit-android #245: 0.12.0 CameraListener is not available

p-canniff, 2017-11-17:
Trying to work in the 0.12.0 updates, but when I switch from 0.11.0 it is not possible to locate or import the CameraListener Class.

I have tried running this with both 26/27 target SDKs and appCompatv 26.+ as well as 27.0.0

If I switch back to 0.11.0 it works fine, but I can't seem to see why the Listener would be missing or not located. Is there a new approach for .setCameraListener in 0.12.0 that I am missing?

Also, it looks like 0.11.0 is having an issue with running something on the mainThread. I receive a "view cannot be access from outside the hierarchy" error when loading it the first time after logging into the app. Was hoping my upgrade to 0.12.0 would solve this.

Also - very interesting that the demo app in Github doesn't have example code for the CameraListener. Did I also miss this?

Lastly, I am anticipating and waiting for the pinchToZoom functionality. Is there an update or when this will be released?

Thanks you,
Philip

calebdre, 2017-11-17:
Just to answer the demo app question:  the demo doesn't look like it takes pictures, but [the documentation](http://docs.camerakit.website/#/?id=capturing-images) has a pretty good example how to to use the `CameraListener`.

p-canniff, 2017-11-18:
I understand how the CameraListener functions. I have been using it in my code for some time. The issue is not understanding, but version compatibility. When I upgrade my sdkBuildTools to 26.0.2 (previous 25.0.3) The cameraListener, and .setCameraListener method were both inaccessible. 

Due to a deadline crunch, I had to revert back a couple version just so I could deploy. Would really like to progress the changes though. Will wait for Zoom to be available before trying again. 

Thanks.

romeroz, 2017-11-19:
Same problem. I just went back to using `implementation 'com.flurgle:camerakit:0.9.17'` and that worked. You can keep all the rest of the code the same just change package where appropriate in xml, java, etc.

jproffitt, 2017-11-20:
I've found the same thing. The docs say to use `CameraListener`, but there is no `CameraListener`. It was removed in this commit: https://github.com/wonderkiln/CameraKit-Android/commit/093778182e8a7d632b9902b9d0dc6679897ef1d5

Do the docs just need to be changed?

p-canniff, 2017-11-20:
Glad I wasn't crazy here. I am currently running `compile 'com.wonderkiln:camerakit:0.10.1'` with target SDK 25. When I was running SDK 26, it showed a missing v26.xml property.

Something wonky here in terms of versioning support.

austinkettner, 2017-11-20:
@p-canniff @jproffitt @romeroz @calebdre 

We will make some refinements to the documentation as part of our 0.13.0 release (November Release). We will also address many varying issues in this release, including this. Can you outline your ideal outcome, given CameraListener and intended use? If that makes sense.

p-canniff, 2017-11-21:
So I would like my code to run with the CameraListener as it always has. If you made a larger change here I can adapt. 

My biggest issue is I can't work in AS 3.0.0 because it has a min. SDK of 26.

jproffitt, 2017-11-21:
@austinkettner It seems you removed `CameraListener` for a reason. I'm ok with that as long as the documentation reflects how to really use the camera now. It still talks about using `CameraListener`. 

So I just need to know what the correct way to do it now is. I've seen some mentions of using something like `@OnCameraKitEvent(CameraKitEvent.class)`, and I've also seen using a new class called `CameraKitEventListener` like here: https://github.com/wonderkiln/CameraKit-Android/issues/250. So what is the proper way of doing it now? The documentation should be updated accordingly.

stealthcopter, 2017-11-27:
I copied the code from this issue to get it working
https://github.com/wonderkiln/CameraKit-Android/issues/250 

    private CameraKitEventListener cameraListener = new CameraKitEventListener() {
        @Override
        public void onEvent(CameraKitEvent cameraKitEvent) {
            switch (cameraKitEvent.getType()) {
                case CameraKitEvent.TYPE_CAMERA_OPEN:
                    canTakePicture = true;
                    break;

                case CameraKitEvent.TYPE_CAMERA_CLOSE:
                    canTakePicture = false;
                    break;
            }
        }

        @Override
        public void onError(CameraKitError cameraKitError) {

        }

        @Override
        public void onImage(CameraKitImage cameraKitImage) {
            cameraKitImage.getJpeg(); // will return byte[]
        }

        @Override
        public void onVideo(CameraKitVideo cameraKitVideo) {

        }
    };

jproffitt, 2017-11-27:
I also used this code which I found from the demo:

    // in onCreate
    mCameraView.bindCameraKitListener(this);

    @OnCameraKitEvent(CameraKitEvent.class)
    public void cameraEvent(CameraKitEvent event) {
        if(event.getType().equals(CameraKitEvent.TYPE_CAMERA_OPEN)) {
            // do open event logic here
        }
    }

    @OnCameraKitEvent(CameraKitImage.class)
    public void imageCaptured(CameraKitImage image) {
        // do on image event logic here
    }

    @OnCameraKitEvent(CameraKitVideo.class)
    public void videoCaptured(CameraKitVideo cameraKitVideo) {
        // do on video event logic here
    }

    @OnCameraKitEvent(CameraKitError.class)
    public void onError(CameraKitError cameraKitError) {
        // do on error event logic here
    }

moses-lee, 2017-12-07:
So which version of camerakit is currently functional for SDK 25?

austinkettner, 2018-03-08:
A WIP in #318 can you please test that out and provide comments as to what you would like to see from a Listener class?

closed 2018-03-08 by austinkettner