CameraKit/camerakit-android #246: Text Recognition via Mobile Vision API

calebdre, 2017-11-17:
### Is this a bug report?
[image]

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?
[image]

It would be awesome to expose an api receiving the current text that's on the screen.
[There barley any documentation for it](https://developers.google.com/vision/text-overview)  but they do provide a [code lab](https://codelabs.developers.google.com/codelabs/mobile-vision-ocr/) for helping people get started, even though they do A LOT of the work for you (at least the code is there to look at.)

I'm checking out the code from the code lab and will be working on this this weekend.

austinkettner, 2017-11-20:
We think this is a great idea and would love to see work toward this. We have some thoughts on different directions we could take to achieve this but would really dig seeing a PR on this front! Would be a fun collaboration effort.

calebdre, 2017-12-03:
#249 merged

closed 2017-12-03 by calebdre

austinkettner, 2017-12-03:
Thanks again for your PR and help here @calebdre very very appreciated!

hardysim, 2018-12-13:
Is this still available (in `1.0.0-beta3.10`)?