CameraKit/camerakit-android #250: setCameraListener method was changed ?

omercsoylu, 2017-11-21:
### Is this a bug report?

Maybe Yes it is

for 0.12 update, setCameraListener function is not there. android studio give an error, " resolved method" as it.

jaksab, 2017-11-21:
Hi!
Change setCameraListener to addCameraKitListener and implement new listener: 

```private CameraKitEventListener cameraListener = new CameraKitEventListener() {
        @Override
        public void onEvent(CameraKitEvent cameraKitEvent) {
            switch (cameraKitEvent.getType()) {
                case CameraKitEvent.TYPE_CAMERA_OPEN:
                    canTakePicture = true;
                    break;

                case CameraKitEvent.TYPE_CAMERA_CLOSE:
                    canTakePicture = false;
                    break;
            }
        }

        @Override
        public void onError(CameraKitError cameraKitError) {

        }

        @Override
        public void onImage(CameraKitImage cameraKitImage) {
            cameraKitImage.getJpeg(); // will return byte[]
        }

        @Override
        public void onVideo(CameraKitVideo cameraKitVideo) {

        }
    };```

omercsoylu, 2017-11-21:
thanks i use this. its working for captureImage. but onVideo not work with captureVideo(). 

i try this code

`cameraView.addCameraKitListener(new CameraKitEventListener() {
           
           @Override
           public void onVideo(CameraKitVideo cameraKitVideo) {
               System.out.println("working?");

           }
       });

       cameraView.captureVideo();`

its not give a result.

watire, 2017-11-24:
change to  
cameraView.captureImage(new CameraKitEventCallback<CameraKitImage>() {
                    @Override
                    public void callback(CameraKitImage event) {
                        byte[] jpeg = event.getJpeg();
                    }
                });

DaCrAn, 2017-12-10:
This is exactly what i need thanks. I think someone should change the Documentation

austinkettner, 2017-12-11:
Will update the documentation soon! Thanks @watire @DaCrAn

brianliyang, 2018-04-24:
Just wanted to bump this and remind the devs to update the documentation. I was upgrading my API from "com.flurgle:camerakit:0.9.17" and although it wasn't hard to find this post, I think new users might be inconvenienced with outdated docs. Thanks for this library!

emersoncloud, 2018-05-02:
@brianliyang I've updated the documentation to reflect the latest image and video capture methods. Let me know if you have any other comments or concerns. Thanks for bringing this to our attention!

closed 2018-05-02 by emersoncloud

hablema, 2018-05-18:
This dosen't give me a result, can anyone tell me what i'm doing wrong. I just updated from com.flurgle.camerakit. The camerakiteventlistener also dosen't get called

```
cameraView.captureVideo(new CameraKitEventCallback<CameraKitVideo>() {
                    @Override
                    public void callback(CameraKitVideo cameraKitVideo) {

                        Log.d("Video","recorded");

                    }
                });
```

I'm using `implementation 'com.wonderkiln:camerakit:0.13.1'`

lights-a5, 2018-07-02:
Am I looking at an old website? This issue says that the documentation was updated, but this still says to use CameraListener. http://docs.camerakit.website/#/?id=camerakit-for-android

emersoncloud, 2018-07-18:
@lights-a5 Yes, the website that you linked is outdated. To avoid any more confusion I am saving documentation changes for one entirely new set in the coming weeks. For now, check the PR notes for basic usage details #318.

I've created a new issue for documentation changes and you can watch for updates there! #374