CameraKit/camerakit-android #252: Getting null byte[] below 5.0 devices and in API - 16 app is getting crash while calling camera.capture()

himangipatel, 2017-11-21:
Here is my gist File : https://gist.github.com/himangipatel/21f17d040e79b84e6071c395c364f345

FATAL EXCEPTION: CameraViewWorker
                                                        java.lang.ExceptionInInitializerError
                                                            at com.wonderkiln.camerakit.PostProcessor.getJpeg(PostProcessor.java:50)
                                                            at com.wonderkiln.camerakit.CameraView$5.imageCaptured(CameraView.java:439)
                                                            at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:288)
                                                            at android.hardware.Camera$EventHandler.handleMessage(Camera.java:734)
                                                            at android.os.Handler.dispatchMessage(Handler.java:99)
                                                            at android.os.Looper.loop(Looper.java:137)
                                                            at android.os.HandlerThread.run(HandlerThread.java:60)
                                                         Caused by: java.lang.UnsatisfiedLinkError: Cannot load library: load_library[1098]: Library '/system/lib/libhoudini.so' not found
                                                            at java.lang.Runtime.loadLibrary(Runtime.java:370)
                                                            at java.lang.System.loadLibrary(System.java:535)
                                                            at com.wonderkiln.camerakit.BitmapOperator.<clinit>(BitmapOperator.java:81)
                                                            at com.wonderkiln.camerakit.PostProcessor.getJpeg(PostProcessor.java:50) 
                                                            at com.wonderkiln.camerakit.CameraView$5.imageCaptured(CameraView.java:439) 
                                                            at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:288) 
                                                            at android.hardware.Camera$EventHandler.handleMessage(Camera.java:734) 
                                                            at android.os.Handler.dispatchMessage(Handler.java:99) 
                                                            at android.os.Looper.loop(Looper.java:137) 
                                                            at android.os.HandlerThread.run(HandlerThread.java:60) 

* Below is my code 
camera.captureImage(new CameraKitEventCallback<CameraKitImage>() {
          @Override public void callback(CameraKitImage cameraKitImage) {
            imgByteArr = cameraKitImage.getJpeg();
            handlePermisssion();
          }
        });

dwillmc, 2017-11-21:
Fixed in `0.12.1`, just stable released.

closed 2017-11-21 by dwillmc

himangipatel, 2017-11-21:
still facing sam issue

himangipatel, 2017-12-05:
**Still getting same error after updating to com.wonderkiln:camerakit:0.12.3 version** 

FATAL EXCEPTION: CameraViewWorker
                                                          java.lang.ExceptionInInitializerError
                                                              at com.wonderkiln.camerakit.PostProcessor.getJpeg(PostProcessor.java:49)
                                                              at com.wonderkiln.camerakit.CameraView$5.imageCaptured(CameraView.java:447)
                                                              at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:302)
                                                              at android.hardware.Camera$EventHandler.handleMessage(Camera.java:815)
                                                              at android.os.Handler.dispatchMessage(Handler.java:99)
                                                              at android.os.Looper.loop(Looper.java:153)
                                                              at android.os.HandlerThread.run(HandlerThread.java:60)
                                                           Caused by: java.lang.UnsatisfiedLinkError: Cannot load library: link_image[1907]:   126 could not load needed library 'libJniYuvOperator.so' for 'libJniBitmapOperator.so' (load_library[1109]: Library 'libJniYuvOperator.so' not found)
                                                              at java.lang.Runtime.loadLibrary(Runtime.java:370)
                                                              at java.lang.System.loadLibrary(System.java:535)