CameraKit/camerakit-android #256: Fatal UnsatisfiedLinkError Crash on 4.1 & 4.2 Devices

jpribble, 2017-11-22:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Huawei Prism II: Android 4.1.1
HTC One X+: Android 4.2.2

### Steps to Reproduce

1. Launch the CameraKit demo app (v0.12.1) on a 4.1/4.2 device.
2. Take a picture.

### Expected Behavior

Picture should be captured without crashing.

### Actual Behavior

Demo app crashed due to a fatal, uncaught UnsatisfiedLinkError.

[Huawei Prism II](https://www.gsmarena.com/t_mobile_prism_ii-5550.php): Android 4.1.1:
> UnsatisfiedLinkError: Cannot load library: link_image[1893]:   132 could not load needed library 'libJniYuvOperator.so' for 'libJniBitmapOperator.so' (load_library[1095]: Library 'libJniYuvOperator.so' not found)
                                                                              at java.lang.Runtime.loadLibrary(Runtime.java:375)
                                                                              at java.lang.System.loadLibrary(System.java:535)
                                                                              at com.wonderkiln.camerakit.BitmapOperator.<clinit>(BitmapOperator.java:81)
                                                                              at com.wonderkiln.camerakit.PostProcessor.getJpeg(PostProcessor.java:50) 
                                                                              at com.wonderkiln.camerakit.CameraView$5.imageCaptured(CameraView.java:439) 
                                                                              at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:288) 
                                                                              at android.hardware.Camera$EventHandler.handleMessage(Camera.java:757) 
                                                                              at android.os.Handler.dispatchMessage(Handler.java:99) 
                                                                              at android.os.Looper.loop(Looper.java:137) 
                                                                              at android.os.HandlerThread.run(HandlerThread.java:60) 

[HTC One X+](https://www.gsmarena.com/htc_one_x+-4976.php): Android 4.2.2:
> UnsatisfiedLinkError: Cannot load library: soinfo_link_image(linker.cpp:1635): could not load library "libJniYuvOperator.so" needed by "libJniBitmapOperator.so"; caused by load_library(linker.cpp:745): library "libJniYuvOperator.so" not found
                                                                              at java.lang.Runtime.loadLibrary(Runtime.java:371)
                                                                              at java.lang.System.loadLibrary(System.java:535)
                                                                              at com.wonderkiln.camerakit.BitmapOperator.<clinit>(BitmapOperator.java:81)
                                                                              at com.wonderkiln.camerakit.PostProcessor.getJpeg(PostProcessor.java:50) 
                                                                              at com.wonderkiln.camerakit.CameraView$5.imageCaptured(CameraView.java:439) 
                                                                              at com.wonderkiln.camerakit.Camera1$2.onPictureTaken(Camera1.java:288) 
                                                                              at android.hardware.Camera$EventHandler.handleMessage(Camera.java:792) 
                                                                              at android.os.Handler.dispatchMessage(Handler.java:99) 
                                                                              at android.os.Looper.loop(Looper.java:158) 
                                                                              at android.os.HandlerThread.run(HandlerThread.java:60)

browningweb, 2017-12-03:
I have this problem on android 4.2.2.

4brunu, 2017-12-12:
I submit a PR with a fix for this issue https://github.com/wonderkiln/CameraKit-Android/pull/279.
While the PR isn't merged, you can do a quick fix by adding this to your `Activity` or `Fragment` where the `com.wonderkiln.camerakit.CameraView` is.

```
    static {
        System.loadLibrary("jpge");
        System.loadLibrary("jpgd");
        System.loadLibrary("JniYuvOperator");
        System.loadLibrary("JniBitmapOperator");
    }
```

dwillmc, 2017-12-12:
Thanks for your PR @4brunu , merged into next release branch. Will release today.

4brunu, 2017-12-12:
This was fast! thanks 👍

closed 2017-12-13 by dwillmc

udanieli, 2018-01-17:
I am testing master (47cda44ff90be5b9e174bc384ee54e04e7cbf080) and am experiencing the _UnsatisfiedLinkError_. I also tried to deploy the demo apk directly to the phone, instead of using the _Run_ function of Android Studio (suspecting some Android Studio flaw). Am I doing something wrong?

4brunu, 2018-01-17:
@udanieli Could show the stack strace?

udanieli, 2018-01-17:
Thank you for the quick repoly. Stack trace is null. I only see this:
`java.lang.UnsatisfiedLinkError: Cannot load library: link_image[1891]: 25444 could not load needed library 'libyuvOperator.so' for 'libjpegTransformer.so' (load_library[1093]: Library 'libyuvOperator.so' not found)`
[image]
here is something that seems a call stack:
[image]
I tried to add --stacktrace --debug parameters to gradle cmdline. Logcat is empty, I have checked _No filters_ option, nothing.
I am a frustrated Android newbie. I have been working for years with Java but never fell in love with it. I still have the clear feeling of not having control over things in the Java ecosystem, especially debugging seems a hard task to accomplish. Please have mercy.

4brunu, 2018-01-17:
Can you share a sample project showing the error?

udanieli, 2018-01-17:
It's the [demo](https://github.com/CameraKit/camerakit-android/tree/master/demo) module in library's project, at revision 47cda44ff90be5b9e174bc384ee54e04e7cbf080 (did not touch any file).

udanieli, 2018-01-18:
Looking to the source code and the commit you pointed me (https://github.com/CameraKit/camerakit-android/pull/293/commits/431f59d45da4e56a367c9ad9ef3dbdf370074807) it seems that this fix should be done again. Just testing, I made this simple patch:

https://github.com/udanieli/camerakit-android/commit/06e55ac8825cd0e5d181bf5a79dc21e977f618ed

then the demo crashes anyway for an _OutOfMemoryException_ thrown from [here](https://github.com/CameraKit/camerakit-android/blob/47cda44ff90be5b9e174bc384ee54e04e7cbf080/demo/src/main/java/com/wonderkiln/camerakit/demo/PreviewActivity.java#L56), maybe because my camera creates a really big photo. Running the demo on an emulated device (640x480) no exception, the demo runs smoothly. I think it's [another story](https://developer.android.com/topic/performance/graphics/load-bitmap.html).

4brunu, 2018-01-18:
@udanieli Your fix looks good 👍 
Yes, thats a separated issue.
Could you please submit a Pull Request with the fix?

austinkettner, 2018-01-19:
Second that, would love a PR, we can look at immediately merging.

PR #280 (2017-12-13): Release v0.13.1
Resolves #256 
Resolves #19 
Resolves #257 
Resolves #282 
Resolves #273