CameraKit/camerakit-android #257: Camera preview is not correctly displayed when changing orientation

Rakatan, 2017-11-23:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Nexus 5X : API 26
Samsung Galaxy S4 : API 21
LG G3: API 21

### Steps to Reproduce

1. Create a new project
2. Add CameraKit version 0.12.1 to your build.gradle file
3. Create an activity
4. Add a CameraView to your main activity layout
5. Start and stop the CameraView in the onResume and onPause methods respectively
6. Run the project
7. Switch the device orientation from portrait to landscape

### Expected Behavior

The preview should be properly sized as per the requirements of the new aspect ratio.
The expected behavior is present in version 0.11.2.

### Actual Behavior

The preview is skewed as if no aspect ratio adjustments were made.
This broken behavior is present in versions 0.12.0, 0.12.1 and 0.13.0

### Reproducible Demo

https://github.com/Rakatan/CameraKitOrientation
Clone the project and simply run it. Changing the device orientation to landscape will reveal the issue.

pennadev, 2017-11-27:
I would like to add that simply starting the activity in landscape also does not work as intended, the preview is not the correct ratio. CameraView is completely unusable in landscape orientation.

criistii, 2017-12-04:
I have the exact same issue. Any fix?

DaCrAn, 2017-12-11:
+1
Same issue here. is there any issue regarding this?

closed 2017-12-13 by dwillmc

DaCrAn, 2017-12-14:
Sorry, I tested this on my Pixel XL android 8.1.0 and still not fixed, when you set in landscape mode when you take a photo the issue back.

commit c1e42885 (2017-12-13): Use display orientation instead of device orientation for layouting preview

Resolves #19 Resolves #257
commit 5875fded (2017-12-13): Fix preview rotation issues

Resolves #257, Resolves #282, Resolves #19
PR #280 (2017-12-13): Release v0.13.1
Resolves #256 
Resolves #19 
Resolves #257 
Resolves #282 
Resolves #273