CameraKit/camerakit-android #259: Image is rotated 180 degree after taking picture with method still and front camera

nguyenvp08, 2017-11-26:
In the demo app, I changed the camera method to "STILL" and switch to front camera, the picture after captured is rotated 180 degree, it always happens with my device (Samsung Galaxy S7 and S8).
Thanks.

gh7800, 2017-11-29:
I to.
China  XiaoMi6  Android7.1

developer--, 2017-11-30:
me too, lenovo 6.0.0

GeoffGodwin, 2018-04-29:
Is it possible that camerakit isn't checking the EXIF orientation? I haven't had time to check yet but in a previous app when I implemented the camera API manually I had to explicitly check the EXIF orientation data from the camera and rotate accordingly.

This Stack Overflow post covers it:
https://stackoverflow.com/questions/14066038/why-does-an-image-captured-using-camera-intent-gets-rotated-on-some-devices-on-a

austinkettner, 2018-04-29:
Hey guys, can you please try the 1.0.0 PR? #318 -- it addresses this issue in a very robust way. If you notice any issues post in the PR as we are getting ready to release a stable version of that.

closed 2018-04-29 by austinkettner