CameraKit/camerakit-android #260: How can I prevent the few seconds "result" after calling cameraKit.captureImage() ?

lsw19921030, 2017-11-27:
I noticed that if I call `cameraKit.captureImage()`,

the cameraView will display a "result" image, and `onImage()` will be called only after the result image is dismissed.

How do I avoid this behavior? My expected behavior is taking a picture will not display a preview, and onImage() is called immediately.

Thanks!

DvlopRyanSpears, 2018-02-02:
Is your problem solved ? i'm facing the same problem too.

siralam, 2018-02-02:
@DvlopRyanSpears Haha although saying this here might be a bit rude, I gave up on this library soon after that.

A similar library would be [this one](https://github.com/natario1/CameraView), But I personally suggest [this one](https://github.com/Fotoapparat/Fotoapparat) the most.

I even wrote [my own one](https://github.com/siralam/CameraViewPlus) by forking Google's CameraView before discovering Fotoapparat, see if it suits your needs.

DvlopRyanSpears, 2018-02-02:
Thanks for your suggestion.

austinkettner, 2018-02-02:
Apologies for the lack of direct involvement on this ticket. We have been hard at work on 1.0.0, and hope to issue a PR very soon for this in hopes of having testers help refine before we issue it. Our focus has been on a lot of these tickets and things like this should be resolved, but if not will be much simpler to directly handle now thanks to many of our changes.

Expect it soon!

emersoncloud, 2018-07-30:
This issue has since been resolved in the latest release of CameraKit 1.0.0 beta1. If you have any more delay issues let us know and we can dig into it further.

closed 2018-07-30 by emersoncloud