CameraKit/camerakit-android #264: is there any option or trick to avoid audio and video permissions? I don't need them

developer--, 2017-11-29:


austinkettner, 2017-11-29:
If you would like to see a feature of explicitly defining 'required' permission invoked, can you please update this ticket to be more clear on expectations and what would be helpful to you? We can then evaluate adding this in!

closed 2017-11-29 by austinkettner

developer--, 2017-11-30:
the library is great and if I need to use it for only app that requires only image capturing, inserting this library causes, that app permissions there is an extra Microphone permission which is useless

candalo, 2017-12-12:
I think that's a good thing. In my case, I use the lib only for image capturing, thereby the audio and video permissions are useless. Is it possible to make these permissions optional? Something like that:

```java
cameraView.setUseOnlyForImageCapturing(true)
```

In this case, will not be necessary audio and video permissions.

candalo, 2017-12-12:
There is already a method for this, I saw in the documentation :)

```java
cameraView.setPermissions(CameraKit.Constants.PERMISSIONS_PICTURE);
```

developer--, 2017-12-12:
yes, but when user goes through app settings, there is an audio permission listed