CameraKit/camerakit-android #265: Video -Path  -Name

gh7800, 2017-11-30:
startvideo:
		mCameraView.captureVideo();
stopvideo:
              mCameraView.stopVideo();
I
Hope that you can customize the name and path

georgeRCluj, 2017-12-13:
@gh7800, did you find out how to effectively capture the video? what method to use? Since the documentation is outdated http://docs.camerakit.website/#/?id=capturing-video

gh7800, 2017-12-14:
ok!

georgeRCluj, 2017-12-15:
ok? that was a question :)

emersoncloud, 2018-06-13:
Hey @gh7800 sorry this issue slipped through the cracks. We have a ticket open for the v1.1.0 milestone to determine the save location. I just added a comment to make sure we add support to set the filename as well. See it here: #336.

emersoncloud, 2018-06-13:
@georgeRCluj Thanks for your interest in CameraKit, and I apologize for the outdated docs. We have been hard at work on release v1.0.0 #318. When the official release is out, we will update all the documentation to match. Expect that in the next couple weeks. Thanks!

closed 2018-07-30 by emersoncloud