CameraKit/camerakit-android #270: NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.setPreviewDisplay(android.view.SurfaceHolder)' on a null object reference

sagartrehan, 2017-12-03:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment
Android All Versions.

Please Note:
I am having a Activity containing a ViewPager. ViewPager contains 3 fragment. I am using FragmentStatePagerAdapter. The Fragment which contains the CameraView is the very first fragment in ViewPager. Might be the combination of ViewPager, ViewPagerStateAdapter and Fragment is causing this issue.

### Steps to Reproduce
Foreground/Background activity containing CameraView

1. Open activity(Lets say ActivityA) which contains fragment. CameraView is the part of Fragment layout.
2. Make sure the ActivityA in foreground. Press the home button to background the app or start another activity on top of ActivityA
3. Repeat steps 2-3 times app crashes.

(Write your steps here:)

1.
2.
3.

### Expected Behavior

App should not crash when activity containing CameraView goes in started/stopped state.

### Actual Behavior

App is crashing

### Reproducible Demo

> Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.setPreviewDisplay(android.view.SurfaceHolder)' on a null object reference
>        at com.wonderkiln.camerakit.Camera1.setupPreview(Camera1.java:500)
>        at com.wonderkiln.camerakit.Camera1.start(Camera1.java:117)
>        at com.wonderkiln.camerakit.CameraView$3.run(CameraView.java:313)
>        at android.os.Handler.handleCallback(Handler.java:742)
>        at android.os.Handler.dispatchMessage(Handler.java:95)
>        at android.os.Looper.loop(Looper.java:154)
>        at android.os.HandlerThread.run(HandlerThread.java:61)

```
package com.orbit.add_post;

import android.app.Activity;
import android.content.Context;
import android.content.Intent;
import android.net.Uri;
import android.os.Bundle;
import android.support.annotation.Nullable;
import android.support.v4.app.Fragment;
import android.view.LayoutInflater;
import android.view.View;
import android.view.ViewGroup;

import com.orbit.R;
import com.orbit.util.UIUtil;
import com.wonderkiln.camerakit.CameraKit;
import com.wonderkiln.camerakit.CameraListener;
import com.wonderkiln.camerakit.CameraView;
import com.yalantis.ucrop.UCrop;

import java.io.File;
import java.io.FileOutputStream;
import java.io.IOException;
import java.util.concurrent.Callable;

import io.reactivex.Single;
import io.reactivex.android.schedulers.AndroidSchedulers;
import io.reactivex.functions.Consumer;
import io.reactivex.schedulers.Schedulers;

public class SelectImageFragment1 extends Fragment {

    private CameraView cameraView;
    private FragmentCallbacks mActivityCallbacks;

    @Override
    public void onAttach(Context context) {
        super.onAttach(context);
        mActivityCallbacks = (FragmentCallbacks) context;
    }

    @Nullable
    @Override
    public View onCreateView(LayoutInflater inflater, @Nullable ViewGroup container, @Nullable Bundle savedInstanceState) {
        return inflater.inflate(R.layout.fragment_camera, container, false);
    }

    @Override
    public void onViewCreated(View view, @Nullable Bundle savedInstanceState) {
        super.onViewCreated(view, savedInstanceState);
        cameraView = view.findViewById(R.id.camera);
        cameraView.setMethod(CameraKit.Constants.METHOD_STANDARD);
        view.findViewById(R.id.close).setOnClickListener(mOnClickListener);
        view.findViewById(R.id.flash).setOnClickListener(mOnClickListener);
        view.findViewById(R.id.rotate).setOnClickListener(mOnClickListener);
        view.findViewById(R.id.capture).setOnClickListener(mOnClickListener);
    }

    @Override
    public void onResume() {
        super.onResume();
        cameraView.start();
    }

    @Override
    public void onPause() {
        cameraView.stop();
        super.onPause();
    }

    private View.OnClickListener mOnClickListener = new View.OnClickListener() {
        @Override
        public void onClick(View v) {
            switch (v.getId()) {
                case R.id.close:
                    getActivity().finish();
                    break;
                case R.id.flash:
                    toggleFlash();
                    break;
                case R.id.rotate:
                    cameraView.toggleFacing();
                    break;
                case R.id.capture:
                    captureImage();
                    break;
                default:
                    break;
            }
        }
    };

    public void toggleFlash() {
        if (cameraView.getFlash() == CameraKit.Constants.FLASH_OFF) {
            cameraView.setFlash(CameraKit.Constants.FLASH_ON);
        } else {
            cameraView.setFlash(CameraKit.Constants.FLASH_OFF);
        }
    }

    private void captureImage() {
        cameraView.setCropOutput(true);
        cameraView.setJpegQuality(70);
        cameraView.setCameraListener(new CameraListener() {
            @Override
            public void onPictureTaken(byte[] picture) {
                super.onPictureTaken(picture);
                mActivityCallbacks.dismissProgressDialog();
                saveCapturedImage(picture)
                        .subscribeOn(Schedulers.io())
                        .observeOn(AndroidSchedulers.mainThread())
                        .subscribe(new Consumer<File>() {
                            @Override
                            public void accept(File s) throws Exception {
                                startCropActivity(s);
                            }
                        }, new Consumer<Throwable>() {
                            @Override
                            public void accept(Throwable throwable) throws Exception {
                                UIUtil.showMessage(getActivity(), "Error");
                            }
                        });
            }
        });
        cameraView.captureImage();
        mActivityCallbacks.showProgressDialog();
    }

    private Single<File> saveCapturedImage(final byte[] image) {
        return Single.fromCallable(new Callable<File>() {
            @Override
            public File call() throws Exception {
                File photo = new File(getActivity().getFilesDir(), "photo.jpg");
                if (photo.exists()) {
                    photo.delete();
                }
                FileOutputStream fos = new FileOutputStream(photo.getPath());
                fos.write(image);
                fos.close();
                return photo;
            }
        });
    }

    private void startCropActivity(File sourcePath) throws IOException {
        Uri sourceUri = Uri.fromFile(sourcePath);
        File destinationFile = new File(getActivity().getApplicationContext().getFilesDir(), String.valueOf(System.currentTimeMillis()));
        if (destinationFile.exists()) {
            destinationFile.delete();
        }
        destinationFile.createNewFile();
        Uri destinationUri = Uri.fromFile(destinationFile);

        UCrop.Options options = new UCrop.Options();
        options.setHideBottomControls(true);
        options.setShowCropGrid(true);
        options.setStatusBarColor(getResources().getColor(R.color.colorPrimary));
        options.setToolbarColor(getResources().getColor(R.color.colorPrimary));

        UCrop.of(sourceUri, destinationUri)
                .withAspectRatio(1, 1)
                .withOptions(options)
                .start(getActivity(), this);
    }

    @Override
    public void onActivityResult(int requestCode, int resultCode, Intent data) {
        if (resultCode == Activity.RESULT_OK && requestCode == UCrop.REQUEST_CROP) {
            final Uri resultUri = UCrop.getOutput(data);
            mActivityCallbacks.onPostImageSelected(resultUri);
        } else if (resultCode == UCrop.RESULT_ERROR) {
            final Throwable cropError = UCrop.getError(data);
            UIUtil.showMessage(getActivity(), cropError);
        }
    }
}

```

(Paste the link to an example project and exact instructions to reproduce the issue.)

austinkettner, 2017-12-03:
Thanks for the detailed report @sagartrehan

sagartrehan, 2017-12-03:
@austinkettner Issue is easily reproducible in Play Store demo app also[One with Black Icon]. Just do 3/4 times foreground/background of demo app while camera visible. Observed on Moto-G1 Android 5.0.

emersoncloud, 2018-07-30:
Thanks again for the issue. This has been resolved in the latest release v1.0.0. Let me know if you are still having issues and I will reopen your ticket!

closed 2018-07-30 by emersoncloud