CameraKit/camerakit-android #272: Why captured bitmap is so big? more than 30MB

developer--, 2017-12-05:
when capture method is standard bitmap size is more than 30 MB tested on two device OS version 5.0 and 6.0

if method is steel bitmap size is about 18 MB.

problem is that it needs about 3-4 second for image capture

xymelon, 2017-12-08:
Because the preview size and picture size is so big. So I change the get preview size method, and return preview size that closest to screen size.

`
            while (descendingSizes.hasNext()) {
                size = descendingSizes.next();
                if (targetRatio == null) {
                    mPreviewSize = size;
                    break;
                }
                if (targetRatio.matches(size)) {
                    //return 'preview size' that closest to screen size.
                    if (size.getWidth() >= CameraKit.Internal.screenHeight
                            && size.getHeight() >= CameraKit.Internal.screenWidth) {
                        mPreviewSize = size;
                    } else {
                        if (mPreviewSize == null) {
                            mPreviewSize = size;
                        }
                        break;
                    }
                }
            }
`

emersoncloud, 2018-07-30:
@xymelon thanks for clearing this up. Closing this issue.

closed 2018-07-30 by emersoncloud