CameraKit/camerakit-android #275: MediaRecorder Stop failed

tsoumalis, 2017-12-07:
### Is this a bug report?
Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Application which simply records a video using CaptureVideo in version 0.13.0

### Steps to Reproduce

There are specific devices where video recording fails, either in one or both cameras. An example of this is Samsung Galaxy S4 (front camera only) and Samsung Galaxy S7 Edge.

Steps:

1. Start capturing using captureVideo()
2. Stop the capturing after a few seconds using stopVideo()
3. You can see the following exception in logcat (and the event for the video file is not fired because the file is null):

Exception java.lang.RuntimeException: stop failed.
android.media.MediaRecorder.stop (MediaRecorder.java)

### Expected Behavior

Expected behavior is for the video file to be saved with no media recorder exception

### Actual Behavior

As described, no video is saved

### Reproducible Demo

I cannot provide the source code for now because it is a part of a commercial project but I am happy to collaborate in order to help solve this strange behavior in a different shared project where the issue is reproduced.

One possible explanation could be the difference in Preview size and Video recording size. I have read on some answers in stack overflow that some devices require both to be the same during recording. 

Another explanation could be that these devices simply need the implementation of Camera2 Api and cannot support video recording with Camera1 no matter what input you give to the media recorder object.

tsoumalis, 2017-12-08:
Setting the preview size of the camera parameters as the recording size solved the issue for Samsung S4. I added the following code in prepareMediaRecorder:

```
private boolean prepareMediaRecorder(File videoFile) throws IOException {
        synchronized (mCameraLock) {
            CamcorderProfile profile = getCamcorderProfile(mVideoQuality);
            mCamera.lock();

            int retries = 0;
            while(retries++ < 20)
                try {
                    mCameraParameters.setPreviewSize(profile.videoFrameWidth, profile.videoFrameHeight);
                    mCameraParameters.setRecordingHint(true);
                    mCamera.setParameters(mCameraParameters);
                    break;
                }
                catch (Exception e)
                {
                    try {
                        Thread.sleep(10);
                    } catch (InterruptedException e1) {
                        e1.printStackTrace();
                    }
                }

            mCamera.unlock();

...
```

I get the camcorder profile before I initialize the media recorder so that I can get the video size and adjust the preview accordingly. The try - catch is because sometimes setParameters fails for no apparent reason so I retry for 20 times and if it keeps failing I continue with some latency of 200 ms.

I also set recording hint to true because some devices might require this value to be so in order to provide the video to the media recorder.

I am waiting for feedback from an S7 user to see if this solves the issue. For anyone else having this problem with a device, please try this workaround and let me know if it works.

tsoumalis, 2017-12-10:
I have noticed that just setting the recording hint to true from camera parameters solves the problem in most devices. I have put this code inside adjustCameraParameters. Maybe there could be a property to specify whether or not you need to record a video and set this value according to this? It could be an XML property.

For the preview/video sizes matching, according to Android documentation (https://developer.android.com/reference/android/hardware/Camera.Parameters.html#getSupportedVideoSizes()):

**getSupportedVideoSizes**

Gets the supported video frame sizes that can be used by MediaRecorder.

**If the returned list is not null**, the returned list will contain at least one Size and one of the sizes in the returned list must be passed to MediaRecorder.setVideoSize() for camcorder application if camera is used as the video source. **In this case, the size of the preview can be different from the resolution of the recorded video** during video recording.

So, maybe we should do a checking for the video sizes and if it is empty then lock the preview size to be equal to recording size. 

There is also a getPreferredPreviewSizeForVideo method, which states that if we are about to record a video then we should not set the preview larger than this returned value.

emersoncloud, 2018-07-30:
Hi @tsoumalis thanks for the issue and suggestion. Currently we are adding video support to the 1.0.0 release. You can check for progress here #318. We will be adding more video features to 1.1.0 and I have created an issue to set preview sizes in video here: #387. Comment there if you have any more feedback!

closed 2018-07-30 by emersoncloud