CameraKit/camerakit-android #278: Pause/Resume features

devoma, 2017-12-09:
Thanks for the great library. I have a question. 

Does your library have features such as pause and continuing video recording? If not, which file can I add. Thank you

tsoumalis, 2017-12-10:
I would recommend that you use the library mp4parser to combine videos that you have recorded. Just be careful when you mix videos that are recorded from front and rear cameras because they have different resolutions and rotations in most devices so they cannot be concated (you can use ffmpeg though to re-encode them with the same settings). 

This method creates the final file almost instantly because it doesn't reencode the videos.

https://github.com/sannies/mp4parser

I personally use this code for concating. Feel free to modify it according to your needs:

`public static boolean appendRecordings(ArrayList<VideoRecording.Recording> files) {
        try {
            List<Movie> inMovies = new LinkedList<Movie>();
            for(int i = 0; i < files.size(); i++) {
                if (!files.get(i).video.exists())
                    continue;
                Movie movie = MovieCreator.build(files.get(i).video.getAbsolutePath());
                inMovies.add(movie);
            }

            List<Track> videoTracks = new LinkedList<Track>();
            List<Track> audioTracks = new LinkedList<Track>();
            for (Movie m : inMovies) {
                for (Track t : m.getTracks()) {
                    if (t.getHandler().equals("soun")) {
                        audioTracks.add(t);
                    }
                    if (t.getHandler().equals("vide")) {
                        videoTracks.add(t);
                    }
                }
            }
            Movie result = new Movie();
            if (audioTracks.size() > 0) {
                result.addTrack(new AppendTrack(audioTracks.toArray(new Track[audioTracks.size()])));
            }
            if (videoTracks.size() > 0) {
                result.addTrack(new AppendTrack(videoTracks.toArray(new Track[videoTracks.size()])));
            }
            Container out = new DefaultMp4Builder().build(result);
            String outputFilePath = Utilities.GetPath("output_append.mp4");
            FileOutputStream fos = new FileOutputStream(new File(outputFilePath));
            out.writeContainer(fos.getChannel());
            fos.close();
        } catch (Exception e) {
            e.printStackTrace();
            return false;
        }

        return true;
    }
`

The recording class is just a class I have created containing information about the recorded file. In this case you only need the file.

tsoumalis, 2018-02-20:
The origin of the problem is the fact that you try to merge 2 videos with different configurations. In most devices, videos from the front and rear camera have a difference of 180 degrees in their rotation metadata.

When merging 2 videos with Mp4 Parser you must be careful that the 2 videos have exactly the same features. This includes resolution, frame rate, DAR and rotation. So, in order to make sure that Mp4 Parser does the job I re-encode all the videos to the same characteristics in case video both from front and rear camera needs to be concated. This is neccessary because some devices might produce videos with totally different characteristics between the 2 cameras and thus cannot be concated instantly.

In order to achieve this I use FFMPEG. I reencode all the videos with some specific standard values (e.g. 720x1280 resolution, 25 fps, dar 16:9) and then use MP4 Parser to combine them in a single final video.

The command I use exactly is the following:
` ffmpeg -y -i video.mp4 -c:v libx264 -preset ultrafast -b:v 3000k -r 25 -vf scale=720:1280,setdar=9:16 -c:a copy video_rotated.mp4`

As you can see I don't specify rotation. That is because FFMPEG removes that metadata and encodes the video without it.

Keep in mind that FFMPEG is only one solution. There might be other solutions for example using the MeidaCodec API.

100nandoo, 2019-03-06:
Does this feature has been implemented?

rakesh0021, 2020-07-16:
is this feature added ?