CameraKit/camerakit-android #283: Can any

georgeRCluj, 2017-12-13:
### Is this a bug report?

I cannot figure out how to record a video. I am using the http://docs.camerakit.website/#/?id=capturing-video which is out-dated.

I have a button and when it starts recording I am writing `layoutUi.camera.captureVideo();` then when pressing button again I am writing `layoutUi.camera.stopVideo();` but when I am using the callback
`layoutUi.camera.captureVideo(new CameraKitEventCallback<CameraKitVideo>() {
                @Override
                public void callback(CameraKitVideo cameraKitVideo) {
                    videoFile = cameraKitVideo.getVideoFile();
                    Toast.makeText(thisContext, "finished recording video", Toast.LENGTH_SHORT).show();
                }
            });`

it is not working, the Toast does not appear. Can anybody help on this one?

boranyldrm, 2017-12-14:
Please look at [this comment on similar issue](https://github.com/wonderkiln/CameraKit-Android/issues/229#issuecomment-344139411). It was helped to me.

georgeRCluj, 2017-12-15:
Permission to storage is granted, but that method does not work. The callback is not called.

georgeRCluj, 2017-12-15:
Great, I found it out finally two days ago and used those overrides ;)
Thanks! ;)

amitsharma492, 2018-02-17:
@georgeRCluj I am facing the same issue. Can you help regarding this?

georgeRCluj, 2018-02-19:
@amitsharma492, did you manage to find a solution? Otherwise I will write you how I did and it worked absolutely great!

austinkettner, 2018-02-19:
@georgeRCluj an example would be appreciated, we can include some additional documentation about this in the documentation site. Additionally, we would appreciate if anyone can test out the 1.0.0 PR for issues like this and comment.

valterh4ck3r, 2018-03-30:
Same issue

Joseph82, 2018-04-10:
Same issue here!

emersoncloud, 2018-07-30:
We are adding more robust video support very soon to 1.0.0. Keep an eye on #318 for more information!

closed 2018-07-30 by emersoncloud