CameraKit/camerakit-android #288: 0.13.1 version cameraview has stretched preview

developer--, 2017-12-15:
### Is this a bug report?

(yes)
# #
version 0.13.0 works fine
OS 6.0.0

4brunu, 2017-12-15:
I have the same problem, but only in a specific device with Android 4.4.2.
Only the preview is stretched, the result image is fine.

DummyCo, 2017-12-15:
Same here: Xiaomi Redmi Note 3, Android 5.02 LRX22G, MIUI 8.5

austinkettner, 2017-12-15:
Ok drafted a 0.13.2 milestone and added this to it, we will see what we can do.

developer--, 2017-12-15:
thanks, and also can anyone fix flash issue? when method mode is steel flashlight does not works #266

austinkettner, 2017-12-15:
Added @developer--

daleksandrovms, 2017-12-29:
I am updated from 0.13.12 - see the same problem. Meizu m5 note

4brunu, 2018-01-05:
For me on one device I don't need to rotate it, it's stretch even in portrait. On the other devices it works, but I only support portrait.
The problem should be https://github.com/wonderkiln/CameraKit-Android/pull/280/commits/c1e42885a2b6b8369482a71874f6cda6d3a1fa57 or https://github.com/wonderkiln/CameraKit-Android/pull/280/commits/5875fded1b7bdac1bcc049fcc3accff4c0ac5cbf
@austinkettner any news on this?

xp-vit, 2018-01-06:
Same reproduced on Nexus 4 (Android 5.1.1) and Nexus 5 (Android 6.0.1). Most likely was broken by:
https://github.com/wonderkiln/CameraKit-Android/commit/5875fded1b7bdac1bcc049fcc3accff4c0ac5cbf

DaCrAn, 2018-01-08:
Actually, this issue is fixed in google cameraview repo. maybe we can apply that fix here.

francescoberardi, 2018-01-16:
Hi,
I have a similar problem also with 0.13.0. In landscape on Samsung galaxy S2 Tablet with this option:
<com.wonderkiln.camerakit.CameraView
        android:id="@+id/camerakit_view"
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:adjustViewBounds="true"
        android:layout_centerInParent="true"
        android:screenOrientation="portrait"
        android:keepScreenOn="true"
        />
I have the preview stretched. Am I making any mistakes?
Thanks

austinkettner, 2018-01-16:
Hey everyone, as we gear up for our 1.0.0 release all of the orientation, stretch, etc tickets have been addressed as we have refactored how we handle all of this. Wanted to drop a not here as this the main reason we are not more directly addressing these issues in 0.13.x or 0.12.x releases. Hope that makes sense! We plan to have `1.0.0-Snapshot` out this week.

Danm72, 2018-01-16:
What branch is that work happening on? Sounds good!

austinkettner, 2018-01-16:
Not yet pushed up for a few reasons, but you will see on a 1.0.0 branch in the next few days! In future releases we don't plan to hold the commits from the repo this long, as it's bad practice, but we had a few reasons this time around.

Danm72, 2018-01-19:
@austinkettner Looking forward to seeing the 1.0.0 snapshot, any way to be notified when its released?

BlueTreeMatt, 2018-01-26:
Any news on this issue? I'm experiencing this on an API Level 17 device

4brunu, 2018-01-26:
There is an open PR with the fix https://github.com/CameraKit/camerakit-android/pull/306
Hope this is merged soon :)

BlueTreeMatt, 2018-01-26:
@4brunu 

Is the fix in #306 not more related to the demo application rather than preview being stretched when calling the CameraKit API directly? I hope I'm wrong!

4brunu, 2018-01-26:
Sorry, you are right. I was making a confusing between this issue and https://github.com/CameraKit/camerakit-android/issues/256#issuecomment-358837297.

dakshays, 2018-01-31:
I'm facing the same issue. Is 1.0.0 snapshot released?

pablofridman, 2018-02-02:
Same error, preview is distorted when screen launch with portrait and then change to landscape.

littleariel, 2018-02-02:
FYI, I had the same issue and reverted back to 0.13.0 as recommended by this thread to fix it. Will stay on this version until 1.0.0 is released.

austinkettner, 2018-02-02:
Hoping to PR 1.0.0 this weekend. Will keep you posted.

Danm72, 2018-02-02:
@littleariel I thought things were okay with 13 but i'm seeing some aspect ratio issues on some tablets, nexus 7/10 for example

austinkettner, 2018-03-08:
Should be resolved in #318 can you please test that out and I will reOpen this if you comment that it is still an issue.

closed 2018-03-08 by austinkettner

maruthivarman, 2020-05-29:
> Hi,
> I have a similar problem also with 0.13.0. In landscape on Samsung galaxy S2 Tablet with this option:
> <com.wonderkiln.camerakit.CameraView
> android:id="@+id/camerakit_view" [...]

How to resolve issue?