CameraKit/camerakit-android #294: Preview is distorted when screen launch with portrait and then change to landscape

tzzen93, 2017-12-24:
The camera is distorted when screen launch with portrait and then change to landscape orientation immediately. At that timing, the camera is distorted. Thank you very much!

xp-vit, 2018-01-06:
I think this is a duplicate of: https://github.com/wonderkiln/CameraKit-Android/issues/288

Danm72, 2018-01-12:
I'm seeing this too.

Danm72, 2018-01-16:
Was forced to lock orientations.

austinkettner, 2018-03-08:
Should be resolved in #318 can you please test that out and I will reOpen this if you comment that it is still an issue.

closed 2018-03-08 by austinkettner

jatin-lakhani, 2018-05-04:
+1 Its still exist with version 0.13.1
Any workaround on this?

emersoncloud, 2018-05-04:
@jatin-lakhani There is not currently a workaround in version 0.13.1. The issue should be resolved in v1.0.0 which is currently in review, #318. Try out 1.0.0 and let me know if you are running into the same issues.

jatin-lakhani, 2018-05-07:
Thank you for the reply! Version 1.0.0 is quite better but video integration seems not integrated yet. Is there any alternative I can use both feature camera capture + video recording?

emersoncloud, 2018-05-13:
@jatin-lakhani Not yet. We expect to have video support in 1.0.0 this week!

fearenales, 2019-07-24:
@emersoncloud sorry for necrobumping this issue, but is video already supported in 1.0.0? If not, any idea when it should be available and when 1.0.0 will be released?