CameraKit/camerakit-android #295: Flash doesnt work on STILL METHOD

pablofridman, 2017-12-27:


Vichy97, 2018-01-11:
I dont think flash is triggered because its just capturing the preview but you can workaround this by setting flash to torch on click then setting it to off on picture callback. It would be nice to check if auto flash would be triggered though as right now there is no way to do that

tzzen93, 2018-01-12:
Me too, Flash light cannot work in during capturing. Runtime exception and app was force closed.

austinkettner, 2018-01-12:
Looking into this, however in our 1.0.0 release this is considered resolved. Once 1.0.0 Snapshot is available I will update this issue so you can try it out!

tzzen93, 2018-01-12:
Oh so sorry, Now it is ok, it is working well. 3day ago, that issue is still happen. now it is ok. Thank you so much!

closed 2018-01-12 by austinkettner

Vichy97, 2018-01-12:
When is 1.0.0 out? And will it include camera2 support?

austinkettner, 2018-01-12:
Sometime next week @Vichy97 (potentially sooner) and yes it has Camera2 support!! =)

Vichy97, 2018-01-12:
nice! thank you @austinkettner !

Xenero, 2018-04-20:
Still no 1.0.0 public release?

bpappin, 2018-09-20:
I think I'm using the snapshot of 1.0.0, but the torch/flash is not working on a Nexus 5X.
I just checked the v1.0.0 branch directly, with the same result.

bpappin, 2018-09-26:
@emersoncloud and how do I get the beta3.9 build? 
I am exceedingly confused buy your release process, and working bransches :)
(have you considered a process like gitflow?)

The XML I'm using is as follows:
```
<com.camerakit.CameraKitView
        android:id="@+id/camera"
        android:layout_width="0dp"
        android:layout_height="0dp"
        android:adjustViewBounds="true"
        app:camera_facing="back"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintHorizontal_bias="0.0"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent"
        app:layout_constraintVertical_bias="0.0">
</com.camerakit.CameraKitView>
```

austinkettner, 2018-09-26:
Hey @bpappin thanks for the added information!

You can get the Beta3.9 build from the v1.0.0 PR/branch or (Bintray/Gradle via incrementing the version to match `beta3.9`) which is the primary source of dev, other branches aren't mentioned or advertised as they aren't meant to be tested directly until merged in v1.0.0.

We do follow a process of gitflow but without the 'develop' branch currently =) -- while everything is turbulent currently due to v1.0.0 taking so long to ship and the need to other working branches, we definitely imagine a more seamless and less confusing operation as we move forward, so we hear you loud and clear!