CameraKit/camerakit-android #296: CameraView covers up SnackBar when Preview is shown

Ninjaman494, 2017-12-27:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Samsung Galaxy S4
API 21 (min API 15)

### Steps to Reproduce

1. Create a Layout with a CameraView in it, set width and length to match_parent
2. call cameraView.start()
3. Create a SnackBar and call show() while the preview is showing. 

This works with startRecordingVideo() as well.
### Expected Behavior

The SnackBar should show up above the CameraView when show() is called

### Actual Behavior

SnackBar doesn't show. I believe CameraView is covering it up. I tried running my code with CameraView's visibility set to invisible, and the SnackBar shows up properly. This seems to be a problem with the new version of CameraKit. When I was using an older version(I think 0.9.18), this bug didn't occur.

BorislavKamenov, 2018-04-01:
Try to put specific value (like 300dp) instead of match_parent of the <CameraView ... />
There is similar bug with statusbar and toolbar. I wanted my CameraView to expand to screen width but that cause CameraView to cover both status bar and toolbar.
If you really want your cameraview to get screen width, get the screenwidth in pixels - 1 (yes, subtract 1px) and put it programatically to LayoutParams.width of the CameraView.
Really weird but it works for me and 1px is not that noticeable.

Kiruel, 2018-07-25:
I have the same problem, I put a button inside the CameraView and I can't click on it. With the preview version com.flurgle:camerakit:0.9.17 i don't have this problem.

emersoncloud, 2018-07-30:
Closing this ticket. If you are still running into issues report back here and we will dig into it!

closed 2018-07-30 by emersoncloud